Xilinx/embeddedsw #47: Issue in example code of cyclic mode + AXI DMA ??

vlsireddy, 2018-06-26:
Hi,

     In the xaxidma_example_sgcyclic_intr.c,

 

The cyclic mode setup of TxDMA [MM2S] does NOT follow the [AXI DMA v7.1 14 PG021 April 4, 2018]

 
"This bit should be set/unset only when the DMA is idle or when not running. Updating this bit while the DMA is running can result
in unexpected behavior."

 

https://github.com/Xilinx/embeddedsw/blob/b2f54f649486ec1435f2f6b6e524d9c9c84f2efc/XilinxProcessorIPLib/drivers/axidma/examples/xaxidma_example_sgcyclic_intr.c

 

Code: -

in Txsetup(), the DMA engine is started

in TxSetup() --> XAxiDma_BdRingStart() --> XAxiDma_StartBdRingHw() --> the bit of XAXIDMA_CR_RUNSTOP_MASK is set in this function

 

and in SendPacket() [a much later function in the code flow], the bit of XAXIDMA_CR_CYCLIC_MASK is set in

XAxiDma_SelectCyclicMode()

 

This goes against the Xilinx official document.

 

Thanks

radheyxilinx, 2018-06-27:
Thanks for reporting the issue. Agree XAXIDMA_CR_CYCLIC_MASK  should be set before DMA operation is started. We look into the issue and fix it.

Thanks,
Radhey

radheyxilinx, 2019-01-07:
Fixed in commit: https://github.com/Xilinx/embeddedsw/commit/aad18313da9b9858d0af4b2dfac9e6d0be3eca9d#diff-0ef7166c7060048f0f5fb8b58e98540b

honorpeter, 2019-07-11:
xaxidma_example_sg_intr.c      Can't  Receive  interrupt both TX and RX.  why?

radheyxilinx, 2019-07-11:
Hi @honorpeter, this issue is created for sg cyclic interrupt example. Looks like you are using xaxidma_example_sg_intr.c which is different. Can you post your question on xilinx forum?