Xilinx/embeddedsw #60: mcap not working

abhisheietk, 2018-09-14:
On running mcap in root I am getting this error. "Unable to get the Register Base"
```bash
root $ ./mcap -x 0x7028
Xilinx MCAP device found
Unable to get the Register Base
```

Please help !!

kedare123, 2018-09-14:
Seems to be vendor Id or Device Id is not getting detected 
can you run the lspci --nn on your host PC and comment the output

m4n1c22, 2019-03-21:
@abhisheietk Can you check you if you have enabled the option for mcap in your block design?
If you have a xDMA block, then there is an option within advanced to enable the PR over PCIe. You need to enable it inorder to make the capability to work. Please let me know how it turns out.

Vae1997, 2020-11-11:
> @abhisheietk Can you check, if you have enabled the option for mcap in your block design?
> If you have a xDMA block, then there is an option within advanced to enable the PR over PCIe. You need to enable it inorder to make the capability to work. Please let me know how it turns out.

Hello，I just met the same error.
But when I double click xdma IP core and choose "advance"，I can't find any message about PR over PCIe.
The board what I used is K7，but when I create KCU1500 project and choose dma IP core, it can be found in advance mode.
So，if MCAP is only support KCU？