Xilinx/embeddedsw #70: lwip202 of Vivado 2018.2 keeps bring ethernet up and down

liubenyuan, 2018-11-20:
I am using Vivado 2018.2 and lwip202, and run the demo of Freertos_echo. Lwip202 was configured to 1000 Mbps and the sprintf messages are logged to UART. The board was MYIR Zturn Lite which has a microchip ethernet chip KSZ9031RN,

This demo cannot run and the following messages are printed continuously at the console,
```
ethernet link up
ethernet link down
ethernet link up
````
How could I fix this issue?

liubenyuan, 2018-11-20:
I have to comment out line 353 and 354,
```C
	//if ((eth_link_status == ETH_LINK_UP) && (!phy_link_status))
	//	eth_link_status = ETH_LINK_DOWN;
```
Then this app can be run under my ethernet chip KSZ9031RN.

Is it due to that this chip can not return proper link status at line 351:
```C
phy_link_status = phy_link_detect(xemacp, phyaddrforemac);
```
when autonegitiation is disabled for chips other than,
```
WARNING: Not a Marvell or TI or Realtek Ethernet PHY. Please verify the initialization sequence.
```

liubenyuan, 2018-11-20:
If the emacps is not the following chips,
```C
			XEmacPs_PhyRead(xemacpsp, phy_addr, PHY_IDENTIFIER_1_REG,
							&phy_reg);
			if ((phy_reg != PHY_MARVELL_IDENTIFIER) &&
				(phy_reg != PHY_TI_IDENTIFIER) &&
				(phy_reg != PHY_REALTEK_IDENTIFIER)) {
				xil_printf("WARNING: Not a Marvell or TI or Realtek Ethernet PHY. Please verify the initialization sequence\r\n");
			}
```
then the code in `phy_link_detect` will return invalid status,
```C
	XEmacPs_PhyRead(xemacp, phy_addr, IEEE_STATUS_REG_OFFSET, &status);
	XEmacPs_PhyRead(xemacp, phy_addr, IEEE_STATUS_REG_OFFSET, &status);

	if (status & IEEE_STAT_LINK_STATUS)
		return 1;
	return 0;
```

Therefore caused ethernet link to be ups and downs ?

harini-katakam, 2018-11-21:
Hi,

In 2018.2, hotplug support was added to lwip. "IEEE_STAT_LINK_STATUS" is common to all PHYs.
Judging from the prints, it looks like the link is unstable.
Lwip driver supports the mentioned PHYs by default. The default init/negotiation for most IEEE compliance PHYs will be similar. Could you please check the microchip PHY datasheet and make any manufacturer specific changes in your adapter?

Regards,
Harini

liubenyuan, 2018-11-21:
I print the result of
```C
phy_link_status = phy_link_detect(xemacp, phyaddrforemac);
```
It reports 0 even if the link is up. And the green led on the ethernet periodically blinks.

I patched the code using the following commits, it adds Microchip driver to lwip202,
[](https://github.com/fpgadeveloper/myd-y7z010-projects/tree/master/EmbeddedSw/ThirdParty/sw_services/lwip202_v1_09)

harini-katakam, 2018-11-21:
Hi,

Thanks but I'm afraid we do not have a plan or any process to accept patches on lwip for supporting different PHYs as it is difficult for us to maintain and test in our tree. 

Regards,
Harini

liubenyuan, 2018-11-21:
Hi,

I think the patch provided by fpgadeveloper is pretty mature for the Microchip KSZ9031 product line. If patched, it solves the problem of continuously bring ethernet up and down. And I can confirm that this problem does not relate to a unstable cable or phy connection.

Best.

Benyuan

liubenyuan, 2018-11-22:
Hi, Harini,

If I set `phy_link_speed` fixed to 1000 Mbps, or 100 Mbps, the hotplug periodically checks for link status, and KSZ9031 returns 1 even if link is up already. If I patched lwip202 with KSZ9031 code, and enable `phy_link_speed` to autonegotiate, then after negotiation, the peridotically checks of `phy_link_detect` returns 0 and there is no ethernet link up or ethernet link down messages.

I do not think this is related to hotplug. I think that the results of `phy_link_detect` may be affected by the initialization procedures of ethernet chips. Thanks for your reply anyway, I will maintain a private branch and move on. I will merge your changes if there is any. If you need any debug information, I am glad to help.

Best.

Benyuan

liubenyuan, 2020-06-19:
I think I should stick to patch Microchip KSZ9031 locally. Thank you for your support anyway.

closed 2020-06-19 by liubenyuan

mecitpmk, 2023-09-24:
Hello, There is a problem with the detech PHY in xemaclite. If you use the product mention here :
https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/Errata/KSZ87xx-Errata-DS80000687C.pdf

You will face with the link up-down continuously if you have those product for example..

The for loop in the "detect_phy_emaclite" function, it will return the wrong PHY adress be careful about that ( For example you have 4 Port in your switch, but you only use PORT1, but because of the for loop started from 31 to 0 , you most probably got the PORT4 PHY addr thats why I said its wrong ).