Xilinx/embeddedsw #73: Improper use of #elif to test if an address it defined (breaks axidma tests in normal config)

btashton, 2018-12-27:
Most of the dma test examples have a block that looks something like this
```
#ifdef XPAR_AXI_7SDDR_0_S_AXI_BASEADDR
#define DDR_BASE_ADDR		XPAR_AXI_7SDDR_0_S_AXI_BASEADDR
#elif XPAR_MIG7SERIES_0_BASEADDR
#define DDR_BASE_ADDR	XPAR_MIG7SERIES_0_BASEADDR
#elif XPAR_MIG_0_BASEADDR
#define DDR_BASE_ADDR	XPAR_MIG_0_BASEADDR
#elif XPAR_PSU_DDR_0_S_AXI_BASEADDR
#define DDR_BASE_ADDR	XPAR_PSU_DDR_0_S_AXI_BASEADDR
#endif
```

The problem is that while the first preprocessor test is using an `#ifdef` the else statements check for a truth instead of an define.  In many cases the base address will be 0 for `XPAR_PSU_DDR_0_S_AXI_BASEADDR` so the build results in a warning and picks a poor choice for the address.  This bug is not just in the dma code, and `#elif` code that checks for an address should be using defined like this `#elif defined(XPAR_PSU_DDR_0_S_AXI_BASEADDR)`

radheyxilinx, 2019-01-07:
Thanks @btashton for reporting the issue. Valid point.  We will work on it and the fix will be available in next release.

radheyxilinx, 2019-01-07:
Looking in the later code - if base address XPAR_PSU_DDR_0_S_AXI_BASEADDR  is 0 then DDR_BASE_ADDR is not defined and in that case nothing is added to derive MEM_BASE_ADDR.
So functionally it's still fine. However we should fix the #elif check to avoid false warning.

#ifndef DDR_BASE_ADDR
#warning CHECK FOR THE VALID DDR ADDRESS IN XPARAMETERS.H, \
			DEFAULT SET TO 0x01000000
#define MEM_BASE_ADDR		0x01000000
#else
#define MEM_BASE_ADDR		(DDR_BASE_ADDR + 0x1000000)
#endif