Xilinx/embeddedsw #75: OpenAMP echo test stops FreeRTOS task scheduler

nvl1109, 2019-01-22:
Pls update the code of https://github.com/Xilinx/embeddedsw/blob/release-2018.3/lib/sw_apps/openamp_echo_test/src/system/freertos/machine/zynqmp_r5/helper.c#L23 app_gic_initialize function to make it compatible with FreeRTOS task scheduler.

dtrussel, 2020-08-07:
Still no news on this? This OpenAMP + FreeRTOS example is kind of pointless if it disables the FreeRTOS task scheduler. Took me quite a bit to figure out why my code based on this example was not working until I realized that both the FreeRTOS port and the example itself are configuring the GIC.

vsrik, 2020-08-08:
hi @nvl1109 , @dtrussel . Sorry for the delay. We will get back.

dtrussel, 2020-08-11:
Well i got it working by changing the `helper.c` file:
*  Change `xInterruptController` from `static` to `extern` since it is already defined in the FreeRTOS port code.
* In the `app_gic_initialize`  then only do a `XScuGic_Connect` of `IPI_IRQ_VECT_ID` to `metal_xlnx_irq_isr`

However I am not very familiar with the GIC / metal API, so I am not sure what side effects it may cause to leave out:
```
	/* Only associate interrupt needed to this CPU */
	for (int_id = 32U; int_id<XSCUGIC_MAX_NUM_INTR_INPUTS;int_id=int_id+4U) {
		target_cpu = XScuGic_DistReadReg(&xInterruptController,
						XSCUGIC_SPI_TARGET_OFFSET_CALC(int_id));
		target_cpu &= ~mask_cpu_id;
		XScuGic_DistWriteReg(&xInterruptController,
					XSCUGIC_SPI_TARGET_OFFSET_CALC(int_id), target_cpu);
	}
	XScuGic_InterruptMaptoCpu(&xInterruptController, XPAR_CPU_ID, IPI_IRQ_VECT_ID);
```
this part and the `Exception` enabling/disabling/handling parts.

clorch, 2021-02-10:
I just stumbled over this bug as well. I can confirm that it takes a while before you find out what is not working. Are there plans to address this issue?
@dtrussel Thank you very much for the fix/workaround!

nbenatar, 2021-02-22:
Hi,

I possilibly have the same problem but didnt understand the fix?  Do I just change the static to extern and comment out the lines shown by @dtrussel ?

Regards

kernelchuk, 2021-02-27:
Calling app_gic_initialize() in freertos helper.c is a bug. The fix is based on @dtrussel 's comments (thank you!)
and schduled in **Xilinx 2021.1** release. Thank you for posting.