Xilinx/embeddedsw #93: Error when writing to 64-bit addresses from QSPI

ntubach, 2019-08-02:
The Rx buffer address location within the FlashMsg[x] struct (RxAddr64bit) is never set.

Value should be set to Destination Address (such as from `u32 XFsbl_Qspi32Copy(u32 SrcAddress, PTRSIZE DestAddress, u32 Length)`)

Effect is that functionalities such as [this call to `XQspiPsu_Setup64BRxDma(const XQspiPsu *InstancePtr, XQspiPsu_Msg *Msg)`](https://github.com/Xilinx/embeddedsw/blob/c12743e5d7b0140804699ea818cae66b8d7b62a5/XilinxProcessorIPLib/drivers/qspipsu/src/xqspipsu.c#L1155) will never be hit properly.

nagasureshkumar, 2019-08-08:
How you are setting 64bit address? using r5 fsbl or 32bit a53 fsbl?
if it is r5 or 32bit a53 fsbl, we will never hit this case.

ntubach, 2019-08-10:
The FSBL is writing to a passthrough PL AXI Region that is 64-bit addressed (thus the dest_addr for the transfer is a 64bit value). The a53 is aarch64 configured and does run into this issue, from personal test scenarios.

nagasureshkumar, 2019-11-15:
Sorry for the late reply.
Still one doubt, the issue you are facing is with 32bit fsbl?
i mean, r5fsbl or 32bit a53fsbl?
you have to set RxAddr64bit if you are using 32bit processor to access 64bit address.
Please refer https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/qspipsu/examples/xqspipsu_generic_flash_polled_64bit_dma_r5_example.c,
and the line https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/qspipsu/examples/xqspipsu_generic_flash_polled_64bit_dma_r5_example.c#L1658.
here we are trying to access 64bit address, using r5