Xilinx/embeddedsw #101: Number of Interrupt Configuration Register

Savannah222, 2019-11-06:
UG1087 (v1.7) has only 5 Interrupt Configuration Registers (ICDICR) (0xC08-0xC18).
On the other hand, Interrupt Configuration Register written in xscugic_hw.h is described as 0xC00-0xCFC as comment of XSCUGIC_INT_CFG_OFFSET, and it can be read that there are many Interrupt Configuration Registers.

Which is the correct information?
5 or more?

Thanks and Regards

mubinsyed, 2019-11-06:
Hi @Savannah222 ,

As per GICv2 specifications 0xC00-0xCFC is reserved for interrupt configuration register. Since Zynq UltraScale+ has 187 interrupt IDs (ref https://www.xilinx.com/support/documentation/user_guides/ug1085-zynq-ultrascale-trm.pdf table 13-1), UG1087 has  specified 12 interrupt configuration registers 0xC00 - 0XC2C  (each register accommodates configuration for 16 interrupt id's 16 x 12 = 192), which are sufficient for 187 interrupts. Configuration registers starting from 0xC30 are reserved , as  Zynq UltraScale+ has only 187 interrupts.

Thanks,
Mubin

Savannah222, 2019-11-06:
Hi Mubin ,

thank you for your reply.
Thanks to the kind explanation, I was able to understand!

Thanks ,
Savannah222

closed 2019-11-06 by Savannah222