Xilinx/embeddedsw #102: XAXICDMA_MAX_TRANSFER_LEN one byte short?

pieska, 2019-11-14:
in xaxicdma_hw.h:74:

#define XAXICDMA_MAX_TRANSFER_LEN	0x7FFFFF  /**< Max length hw supports */

I think it should be 0x7FFFFF+1 because it's a length, not an offset. Using 0x7FFFFF as length in XAxiCdma_SimpleTransfer() I'm missing one byte.

radheyxilinx, 2019-11-15:
Hi @pieska , looking into CDMA PG(v4.0) BTT is 23 bits (22 to 0).   A maximum of 8,388,607 (0x7FFFFF) bytes of data can be specified by this field for the associated transfer.  So this seems right.  Please let me know if there any followup queries.

pieska, 2019-11-15:
Thanks a lot, i was misled by the documenation in xaxicdma.c:

 *		- XST_INVALID_PARAM if:
 *		Length out of valid range [1:8M]

8M, not 8M-1. what about 4.1? The limit there is 26bit, 64M-1 (PG034 pg. 40), but the driver has the hard limit of 8M-1.

radheyxilinx, 2019-11-16:
Thanks for the confirmation. Yes, the comment description need to be corrected. 
Related to increased 26bit length support, it's in our plan and will be added in next release. 
I will close this issue request.

pieska, 2019-11-16:
OK, thank you.

closed 2019-11-16 by pieska