Xilinx/embeddedsw #103: Xil_DCacheInvalidateRange() does not invalidate last cache line?

pieska, 2019-11-14:
After transferring 0x800000 of data from 0x10000000 to 0x12000000 with

XAxiCdma_SimpleTransfer(InstancePtr, 0x10000000, 0x12000000, 0x800000,...);

and invalidating the DCache with

Xil_DCacheInvalidateRange(0x12000000, 0x800000)

i read the correct values from 0x127FFFE0 to 0x127FFFFF.

If i just transfer 0x7FFFFF bytes, i do not read the correct values starting from 0x127FFFE0.
If i just invalidate length+1 the read back data is correct. it seems, the last cache line is not properly invalidated .

mubinsyed, 2019-11-15:
Hi @pieska,

Start address and length passed to Xil_DCacheInvalidateRange should be aligned with cache line length( in your case it seems to be 32 bytes).

Assuming cache line length of 32 bytes,

Actual end address taken by   
Xil_DCacheInvalidateRange =  (0x7FFFFF  + 0x12000000) &  ‭0xFFFFFFE0= 127FFFE0

As length requested by you ( 0x7FFFFF ) is not aligned with cache line length (32 bytes), Xil_DCacheInvalidateRange not validating last 31 bytes ( 0x7FFFFF mod 32).

Please let me know in case of follow up queries.

pieska, 2019-11-15:
Thanks a lot for the explanation, I will take care of the alignment but I personally would prefer the function to fail in case of  misalignment. Seems better than not invalidating some data and causing wrong data to be read back.