Xilinx/embeddedsw #127: Xuartlite standalone drivers misses the errors

raczben, 2020-05-22:
The driver reads the Xuartlite's status register more times (like [here](https://github.com/Xilinx/embeddedsw/blob/65c849ed46c88c67457e1fc742744f96db968ff1/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.c#L308)), but it does not process the error values (overrun, framing, parity) in it just in [one place](https://github.com/Xilinx/embeddedsw/blob/65c849ed46c88c67457e1fc742744f96db968ff1/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.c#L612) (at the end of the XUartLite_ReceiveBuffer function). And since the errors are "clear-on-read" type bits, there is no chance to get the errors using the drivers.

I have started a [thread ](https://forums.xilinx.com/t5/Embedded-Development-Tools/Xuartlite-standalone-drivers-misses-the-errors/td-p/1109396) in this topic on Xilinx forum as well.

RavitejaNr, 2020-06-04:
This patch is not needed. 

This patch is adding a new wrapper function to read the status register. The new function “XUartLite_GetStatusReg2” is reading the status register and also calling “XUartLite_UpdateStats”  function which is updating the “**Rx** errors in stats” (**Nothing related to Tx**) structure in the Instance (refer below).  So, there is no need to call "XUartLite_UpdateStats" in the Tx path. Please find the patch and comments in line.

#define XUartLite_UpdateStats(InstancePtr, StatusRegister)  \
{                                                                                                             \
              if ((StatusRegister) & XUL_SR_OVERRUN_ERROR)                           \
              {                                                                                               \
                           (InstancePtr)->Stats.ReceiveOverrunErrors++; \
              }                                                                                               \
              if ((StatusRegister) & XUL_SR_PARITY_ERROR)                   \
              {                                                                                               \
                           (InstancePtr)->Stats.ReceiveParityErrors++;              \
              }                                                                                               \
              if ((StatusRegister) & XUL_SR_FRAMING_ERROR)                           \
              {                                                                                               \
                           (InstancePtr)->Stats.ReceiveFramingErrors++; \
              }                                                                                               \
}

-------------- EXTERNAL PATCH ---------------
From 3bcf727d8e3c016e7f1bdb141fdc41a93847f254 Mon Sep 17 00:00:00 2001
From: Benedek Racz <benedek.racz@propatria-inc.com>
Date: Fri, 22 May 2020 11:58:49 +0200
Subject: [PATCH] fix #127

fix: xuartlite standalone drivers misses the errors
---
.../drivers/uartlite/src/xuartlite.c          | 39 ++++++++++++++-----
.../drivers/uartlite/src/xuartlite.h          |  2 +
2 files changed, 32 insertions(+), 9 deletions(-)

diff --git a/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.c b/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.c
index 13f2067874..95c9fdbaee 100644
--- a/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.c
+++ b/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.c
@@ -93,6 +93,29 @@

 static void StubHandler(void *CallBackRef, unsigned int ByteCount);

+
+/****************************************************************************/
+/**
+*
+* This is a wrapper around the low-level XUartLite_GetStatusReg. This reads
+* the status register with the XUartLite_GetStatusReg, then updates the stats
+* with its value. Do not use the low-level XUartLite_GetStatusReg, because you
+* will miss some errors.
+*
+* @param      InstancePtr is a pointer to the XUartLite instance.
+*
+* @return     A 32-bit value representing the contents of the status register.
+*
+* @note               -
+*
+*****************************************************************************/
+u32 XUartLite_GetStatusReg2(XUartLite *InstancePtr)
+{
+    u32 statusRegister = XUartLite_GetStatusReg(InstancePtr->RegBaseAddress);
+    XUartLite_UpdateStats(InstancePtr, statusRegister);
+    return statusRegister;
+}
+
/****************************************************************************/
/**
*
@@ -222,7 +245,7 @@ unsigned int XUartLite_Send(XUartLite *InstancePtr, u8 *DataBufferPtr,
        * Enter a critical region by disabling the UART interrupts to allow
        * this call to stop a previous operation that may be interrupt driven.
        */
[... 60 lines cut ...]
               * If there is data ready to be removed, then put the next byte
@@ -608,8 +631,6 @@ unsigned int XUartLite_ReceiveBuffer(XUartLite *InstancePtr)
                      InstancePtr->ReceiveBuffer.NextBytePtr[ReceivedCount++]=
                              XUartLite_ReadReg(InstancePtr->RegBaseAddress,
                                                    XUL_RX_FIFO_OFFSET);
-
-                      XUartLite_UpdateStats(InstancePtr, StatusRegister);

**>	We can update the Rx errors stats only after receiving any data (which is already taken care in the existing driver above).**
              }

               /*
diff --git a/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.h b/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.h
index f14aef7bbe..ab4ece1bdb 100644
--- a/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.h
+++ b/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.h
@@ -234,6 +234,8 @@ typedef struct {

 /************************** Function Prototypes *****************************/

+u32 XUartLite_GetStatusReg2(XUartLite *InstancePtr);
+
/*
  * Initialization functions in xuartlite_sinit.c
  */

raczben, 2020-06-05:
Hi @RavitejaNr 

First of all, can you edit your post? Please use the `code` markdown feature, to improve readability.

This patch is needed if you want to detect all errors. This patch is needed in both TX and RX paths. Yes, it is needed in TX. The UART itself is full-duplex, asynchronous communication. This means that bytes can arrive (on the RX side), while we send a message (on the TX side).* Therefore any RX errors can occur at any point in time, during transmitting data as well.

*Yes, this (full-duplex behavior) can be disabled from the higher level of the communication (ie. in higher OSI layers), but the UART itself cannot assume this limitation, and the UART itself must support its full-duplex behavior.

RavitejaNr, 2020-06-08:
Can we have a more descriptive name for the new function(XUartLite_GetStatusReg2) like
"XUartLite_Update_GetStatus" or something which gives a hint of  what we are doing inside?

raczben, 2020-06-11:
Yes, feel free to rename it, I just shared mine quick fix.
I will be happy, this issue will be solved in any way.

raczben, 2020-11-10:
Hi,

The issue still exists in [the latest release.](https://github.com/Xilinx/embeddedsw/blob/df5ef17fb29640e830d954be9dae3710843df157/XilinxProcessorIPLib/drivers/uartlite/src/xuartlite.c)

vsrik, 2020-11-12:
Hi - We somehow could not get into the latest release.  We will get it into the next release , we will get an Answer Record Created.