Xilinx/embeddedsw #133: Swapped device IDs in xttcps_intr_example.c

raczben, 2020-06-23:
[xttcps_intr_example.c][1] has two `SettingsTable` structure: on for *ticker timer* and another for the *PWM timer*, however they are swapped.

```C
#define TTC_TICK_DEVICE_ID	XPAR_XTTCPS_1_DEVICE_ID            //  == 1 !!!!!!!!!!!!!!!!
#define TTC_TICK_INTR_ID	XPAR_XTTCPS_1_INTR

#define TTC_PWM_DEVICE_ID	XPAR_XTTCPS_0_DEVICE_ID            //  == 0   !!!!!!!!!!!!!!!!
#define TTC_PWM_INTR_ID		XPAR_XTTCPS_0_INTR
#define TTCPS_CLOCK_HZ		XPAR_XTTCPS_0_CLOCK_HZ

#define INTC_DEVICE_ID		XPAR_SCUGIC_SINGLE_DEVICE_ID
```

```C
static TmrCntrSetup SettingsTable[NUM_DEVICES] = {
	{100, 0, 0, 0},	/* Ticker timer counter initial setup, only output freq */    // This is the PWM timer   !!!!!!!!!!!!!!!!
	{200, 0, 0, 0}, /* PWM timer counter initial setup, only output freq */       // This is the PWM timer   !!!!!!!!!!!!!!!!
};
```

[1]: https://github.com/Xilinx/embeddedsw/blob/f975a0ea90930587ab55e4d920a31950bd373fa8/XilinxProcessorIPLib/drivers/ttcps/examples/xttcps_intr_example.c

mubinsyed, 2020-11-16:
Hi @raczben ,

Thanks, It would be fixed in 2020.2 release.

sivadur, 2020-11-30:
This is fixed now https://github.com/Xilinx/embeddedsw/commit/b64a9f8d6929b6876b6b8fcefb68bc277b82dd32#diff-5716436d2d0215343a713b5ec2766489ffab52df67781d88f5dc27f72fac3bd6

raczben, 2020-11-30:
Ok, Thanks, we can close this issue.