Xilinx/embeddedsw #137: XIicPs_MasterSendPolled blocks forever when the slave is not present

IFXMatthias, 2020-08-06:
**Steps to Reproduce**

1. Physically remove the device that is supposed to be the I2C slave
2. Call XIicPs_MasterSendPolled to interact with the removed device from above

**Actual Behavior**
* The firmware hangs forever

**Expected Behavior**
* An error code is returned

**Note**
XIicPs_MasterSendPolled  seems to hang in this while loop:

	/*
	 * Check for completion of transfer.
	 */
	while ((IntrStatusReg & XIICPS_IXR_COMP_MASK) != XIICPS_IXR_COMP_MASK){

		IntrStatusReg = XIicPs_ReadReg(BaseAddr, XIICPS_ISR_OFFSET);
		/*
		 * If there is an error, tell the caller.
		 */
		if ((IntrStatusReg & Intrs) != 0U) {
			break;
		}
	}

RavitejaNr, 2020-11-30:
Thanks for reporting, we will fix this in our next release.

dnygren, 2022-05-23:
I see in xiicps_master.c
```
* Ver   Who  Date     Changes
* ----- ---  -------- ---------------------------------------------
* 3.13  rna 11/24/20 Added timeout to XIicPs_MasterSendPolled function.
```
So this issue is now closed?

anirudha1977, 2022-05-24:
Yes. It has been fixed.

dnygren, 2022-05-24:
Thanks for the response. I suggest changing the state of this issue from "Open" to "Closed" so customers know there is a fix available. If I had known there was already a fix in place, it would have saved me some effort looking for a workaround myself.

Abhayakarmach, 2025-07-08:
i dont seee the fix