Xilinx/embeddedsw #138: XPm_GetOpCharacteristic missing units

dtrussel, 2020-08-13:
With the `XPm_GetOpCharacteristic` I am able to query different operating characteristics as `u32` values, but the documentation does not mention anything about the measurement units of the returned values.

E.g. if I query the power characteristic is the returned value in milliwatts?

realarunbala, 2020-08-14:
Hi David, We will get back on details. A quick read of server code tells us that you can use this for wake-up latency, power and temp. The latency is in usec and temp is in Q8.7 format and can be converted to celsius. Power is not supported.

dtrussel, 2020-08-19:
What do you mean by `you can use this for wake-up latency, power and temp` and then `Power is not supported.`?
The Power Management Framework User Guide: For Zynq UltraScale+ MPSoC Devices (UG1199) says:

> type: Type of operating characteristic requested:
> * power: Current power consumption.
> * latency: Current latency in μs to return to active state.
> * temperature: Current temperature. (Unsupported)

jollysxilinx, 2020-08-28:
Hi David,

Power value returns hard coded dummy values calculated by firmware and doesn't have any significance in terms of actual power consumption. We will make sure to update User guide to reflect same information.

Below are details for each parameter:

**Power**
Zynqmp: Power value returned is not actual power consumption of device. It is default dummy power value which is fixed in firmware.
Versal: Not supported

**Latency**
Zynqmp and Versal: Current latency in micro seconds to return to active state.

**Temperature**
Zynqmp: Not Supported
Versal:  current temperature in Celsius (Q8.7 format))