Xilinx/embeddedsw #142: MIN and MAX macros are not guarded by #ifndef in xrfdc.h

ayushrai, 2020-09-24:
The file xrfdc.h defines MIN and MAX macros without a guard condition. If the macros have been defined in some other header then it causes the compiler to issue a warning.

The macro definitions should be guarded by ifndef blocks so that they are defined only if necessary.

keryell, 2020-09-24:
And what happens if
```
#define MAX 42
#define MIN 420
```
?

Macros should be avoided at the first place if possible...

ayushrai, 2020-09-25:
If you want a specific implementation of MIN and MAX, then you first undefine them if they are defined and then start redefining them.

MIN and MAX are found in many different headers. If you do need to use such macros, better use a name that won't conflict with anything else. For example, rename MIN and MAX to - XLNX_MIN and XLNX_MAX. Then use those names in your code.

keryell, 2020-09-26:
My point is that if this header has to be used (i.e. included) by a customer, this is irresponsible programming style.
So why not adapting your suggestion and renaming for example these `MIN` and `MAX` to `__XLNX_MIN`, `__XLNX_MAX` or `__XRFDC_MIN`, `__XRFDC_MAX` or whatever which cannot conflict with any user code according to ISO C standard?

ayushrai, 2020-09-26:
I agree with you on this. I just didn't think of this alternative before. Since we don't have namespaces in C, prefixes can help us achieve similar objective. It is indeed better than using common names like MIN and MAX, unless that is necessary for reasons I can't think of.

For the C++ standard at least, we should not use symbol names starting with an underscore -

"Each name that contains a double underscore __ or begins with an underscore followed by an uppercase letter is reserved to the implementation for any use."

keryell, 2020-09-26:
> "Each name that contains a double underscore __ or begins with an underscore followed by an uppercase letter is reserved to the implementation for any use."

Yes, this is exactly why I am suggesting this. Because a normal user will not use it. :-)

conallo, 2020-09-28:
Hi ayushrai & keryell,
Thanks for opening the thread and commenting.
Yes this was a shortsighted, silly mistake - I will fix this for next release by changing the macros to XRFDC_MAX XRFDC_MIN.
Apologies for this.
Conall.

ayushrai, 2020-09-28:
Hi Conall,

That is a good solution. Thanks for not brushing this (minor) issue off.

I will leave it up to you to decide when to close the issue. 

Regards,
Ayush