Xilinx/embeddedsw #144: zynq_fsbl Makefile builds wrong ps7_init.c

s09bQ5, 2020-10-24:
The Makefile in the zynq_fsbl/src directory first compiles all files in that directory and then copies ps7_init.* from the board directory before it links the executable. So the ps7_init.c file from the board directory will not be compiled and the wrong code will be linked.

First of all the ps7_init.* files from the src directory should be removed. It is unclear which board they are for.
Then one of the following two solutions should be used:
 - Either add $(BSP_DIR)/$(BOARD)/ps7_init.c to the c_SOURCES variable and make sure its compilation will find ps7_init.h.
 - Or add a rule for ps7_init.c that copies it from the board directory and add ps7_init.c to c_SOURCES if the wildcard didn't already add it.

Btw., having to add -O0 to the compiler flags is a sign of bad code quality. Good code also works with -O2.

bsvikram, 2020-11-30:
The issue related to "wrong ps7_init.c" files has been fixed in 2020.2 release.
Commit id:
https://gitenterprise.xilinx.com/embeddedsw/embeddedsw/commit/e97d4c2945c51518256327605f1d0ef396f653ba

Regarding compiler flags:
Vitis supports -O0 for Zynq FSBL and hence Makefile in zynq_fsbl/src also supports -O0. -O0 is not a hard requirement and Zynq FSBL should also work with -O2. However the default setting is -O0.

s09bQ5, 2020-11-30:
Well, you should push more often. You fixed it in July and the fix still was not visible in Github in October.

closed 2020-11-30 by s09bQ5