Xilinx/embeddedsw #148: Possible read eye calculation issue on Vitis 2020.2 release for zynqmp_dram_test?

lukipedio, 2020-12-09:
We get the following result from a zynqmp_dram_test on our board:

[image]

Lane-4 has only a **single number different than zero**. But its eye openness percentage is less than lane-5, for instance:

[image]

Is this behavior expected?

vsrik, 2020-12-10:
Hi ,

We will take a look and get back.

naranimanish, 2020-12-10:
Hi @lukipedio -

Thanks for bringing this up.

The eye width here is correct. The Read QS delay for Lane-4 is 26 (See 'AUTO CENTER' section in the results), which means it cannot go beyond '-26' on left eye for lane-4. The left eye ends there, but here the memory tests do not generate the errors even for -26. The code is written in such a way that it skips modifying the Read QS delay once it reaches 0 ('26' + '-26') for further iterations on the left eye and considers '-26' as the left eye edge.

So the left eye edge for lane-4 is '-26' in your case. We understand this can be better visible by printing something else in case of such scenario. It will be taken care in next version of ZynqMP DRAM Tests application.

We will also attach the patch here soon for your convenience.

Thanks again!

Regards,
Manish

lukipedio, 2020-12-11:
Thanks @naranimanish 

So I can trust the net result of Eye Width.

I'll integrate your patch as soon as it will be available.

lukipedio, 2021-04-22:
Any update on DRAM test reposrts with Vitis 2020.2.

We still notice that the eye width is higher than 416 ns, which is our half clock period (1200 MHz clock set for PS DRAM).

lukipedio, 2021-04-23:
Here you find two DRAM test results on the same hardware (same custom board).
I would say that DRA; test generated with Vitis 2020.2 has problems.

[ddr4_ps_17_17_17_ALLIANCE_2018.2_AR72992.txt](https://github.com/Xilinx/embeddedsw/files/6365833/ddr4_ps_17_17_17_ALLIANCE_2018.2_AR72992.txt)

[ddr4_ps_17_17_17_ALLIANCE.txt](https://github.com/Xilinx/embeddedsw/files/6365825/ddr4_ps_17_17_17_ALLIANCE.txt)
(https://github.com/Xilinx/embeddedsw/files/6365824/ddr4_ps_17_17_17_ALLIANCE_2020.2.txt)

naranimanish, 2021-04-29:
Hi @lukipedio - The fix will be part of Vitis 2021.1 release. The AR for 2020.2 will be available soon. Will keep updating you on the same.

lukipedio, 2021-04-30:
Hi @naranimanish ,
  Thanks.

  Will AR contains a patch? Please let me know.