Xilinx/embeddedsw #153: axidma.h missing parenthesis in

raiatea, 2021-01-07:
#define   XAxiDma_IntrGetEnabled(InstancePtr, Direction)	\
XAxiDma_ReadReg((InstancePtr)->RegBase + \

should have a parenthesis

#define   XAxiDma_IntrGetEnabled(InstancePtr, Direction)	\
(XAxiDma_ReadReg((InstancePtr)->RegBase + \

radheyxilinx, 2021-01-08:
Thanks for reporting the issue. Regarding solution- i think we don't need closing parenthesis in XAxiDma_ReadReg second argument. We will fix this and it will be available in next release. 

--- a/XilinxProcessorIPLib/drivers/axidma/src/xaxidma.h
+++ b/XilinxProcessorIPLib/drivers/axidma/src/xaxidma.h
@@ -644,7 +644,7 @@ typedef struct {
  *****************************************************************************/
 #define   XAxiDma_IntrGetEnabled(InstancePtr, Direction)       \
                        XAxiDma_ReadReg((InstancePtr)->RegBase + \
-                       (XAXIDMA_RX_OFFSET * Direction), XAXIDMA_CR_OFFSET) &\
+                       (XAXIDMA_RX_OFFSET * Direction), XAXIDMA_CR_OFFSET &\
                                                        XAXIDMA_IRQ_ALL_MASK)

closed 2021-01-08 by raiatea

raiatea, 2021-01-11:
Hi,

I'm not sure about your solution regarding the parenthesis after XAXIDMA_CR_OFFSET.

In xaxidma.h, XAxiDma_IntrGetEnabled and XAxiDma_IntrGetEnabled looks alike except the offset,
XAxiDma_IntrGetEnabled uses XAXIDMA_CR_OFFSET, and 
XAxiDma_IntrGetIrq uses XAXIDMA_SR_OFFSET.
In xaxidma.h, XAxiDma_IntrGetEnabled is defined as
#define   XAxiDma_IntrGetEnabled(InstancePtr, Direction)    \
            XAxiDma_ReadReg((InstancePtr)->RegBase + \
            (XAXIDMA_RX_OFFSET * Direction), XAXIDMA_CR_OFFSET) &\
                            XAXIDMA_IRQ_ALL_MASK)

In xaxidma.h, XAxiDma_IntrGetIrq is defined as
#define  XAxiDma_IntrGetIrq(InstancePtr, Direction)    \
            (XAxiDma_ReadReg((InstancePtr)->RegBase + \
            (XAXIDMA_RX_OFFSET * Direction), XAXIDMA_SR_OFFSET) &\
                            XAXIDMA_IRQ_ALL_MASK)

In xaxidma_hw.h, XAxiDma_ReadReg is defined as:
#define XAxiDma_ReadReg(BaseAddress, RegOffset)             \
    XAxiDma_In32((BaseAddress) + (RegOffset))

In xaxidma.h, I suppose XAxiDma_IntrGetEnabled should have the same format as XAxiDma_IntrGetIrq.
That is, a call to XAxiDma_ReadReg(arg1, arg2) and an AND with a mask, all is parenthesis:
( XAxiDma_ReadReg(arg1, arg2) & XAXIDMA_IRQ_ALL_MASK )

That is
    (
         XAxiDma_ReadReg
        (
        (InstancePtr)->RegBase + (XAXIDMA_RX_OFFSET * Direction),
        XAXIDMA_CR_OFFSET
    )
        &
        XAXIDMA_IRQ_ALL_MASK
    )

Hence, my proposal is
#define  XAxiDma_IntrGetEnabled(InstancePtr, Direction)    \
            (XAxiDma_ReadReg((InstancePtr)->RegBase + \
            (XAXIDMA_RX_OFFSET * Direction), XAXIDMA_CR_OFFSET) &\
                            XAXIDMA_IRQ_ALL_MASK)

Best regards

radheyxilinx, 2021-01-11:
That is correct. The solution is same as your proposal.

 #define   XAxiDma_IntrGetEnabled(InstancePtr, Direction)	\
-			XAxiDma_ReadReg((InstancePtr)->RegBase + \
+			(XAxiDma_ReadReg((InstancePtr)->RegBase + \
 			(XAXIDMA_RX_OFFSET * Direction), XAXIDMA_CR_OFFSET) &\
 							XAXIDMA_IRQ_ALL_MASK)

radheyxilinx, 2021-01-11:
Somehow -/- symbols are shows as bullets. The first bullet is original code and second one show modification. 
Hope it is clear.