Xilinx/embeddedsw #167: Driver for AXI HBICAP?

SolitaryThinker, 2021-06-08:
Hi, 

Will the driver for AXI HBICAP be added to this repo? If not, can someone please point to where I can find it. I'm using vitis+vivado to work with a vcu118

Thanks

zyue955, 2024-03-28:
Hi,
I recently have same situation like you, do you find the driver of HBICAP? If so, could you please share me the material? Or at least, I wonder how could I send data to the write FIFO of HBICAP in vitis, is Xil_Out32 ? Looking forward to your reply.

Thank you very much in advance.

anirudha1977, 2024-04-01:
Hi,
Have you looked at: https://github.com/Xilinx/embeddedsw/tree/master/XilinxProcessorIPLib/drivers/hwicap ?

regards

zyue955, 2024-04-01:
> Hi, Have you looked at: https://github.com/Xilinx/embeddedsw/tree/master/XilinxProcessorIPLib/drivers/hwicap ?
> 
> regards

Hi,

Thank you for your reply. Yes, I have looked at this driver which actually is used for HWICAP, and I have tried to use HWICAP to replace HBICAP. I can use the example driver to read IDCODE from FPGA, but when I tried to input my own bitstream data to ICAP usign HWICAP, it always lost response, and after reading the read FIFO, I only got FFFFFFD9 which actually represents CRC error, I don't know the reason. Do you have any solution for this?

Thank you very much!

anirudha1977, 2024-04-01:
Hi,
On cross-checking the HBICAP spec and HWICAP implementation, I understand that, the HWICAP driver will not work for HBICAP IP. Unfortunately, we dont have any driver in our embeddedsw repo that supports this IP.

We will investigate further and get back.

regards