Xilinx/embeddedsw #186: XCanFd_GetFreeBuffer error return value is a valid value

PSheppard-Isotropic, 2022-02-16:
XCanFd_GetFreeBuffer performs a sweep through the buffer ready register (XCANFD_TRR_OFFSET) and returns XST_NOBUFFER (33L) if the end of the register bits is reached with no bits set to 0.

0x33 is a valid return value - that would be buffers 0 and 5 are filled, the rest empty.

PSheppard-Isotropic, 2022-03-30:
RegVal is the contents of XCANFD_TRR_OFFSET, which is the bit mapping of the 32 buffers indicating their busy state.

If each bit in that 32-bit variable is set (all buffers busy), XST_NOBUFFER (33 decimal, not hex) is returned, otherwise RegVal is returned.

i.e. The bit mapping is returned, not a buffer number. 33 (0x21 hex) is a valid bit mapping.

This in CanFd v2.4:
`u32 XCanFd_GetFreeBuffer(XCanFd *InstancePtr)
{

	u32 RegVal;
	u32 Index = 0;

	Xil_AssertNonvoid(InstancePtr != NULL);
	Xil_AssertNonvoid(InstancePtr->IsReady == XIL_COMPONENT_IS_READY);

	RegVal = XCanFd_ReadReg(InstancePtr->CanFdConfig.BaseAddress, XCANFD_TRR_OFFSET);
	while(RegVal & (1 << Index)) {
		Index++;
		if(Index == MAX_BUFFER_INDEX)
			break;

	}
	if(Index == MAX_BUFFER_INDEX)
		return XST_NOBUFFER;

	return RegVal;
}`

sneeli-git, 2022-03-30:
Please check below comment section once https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/canfd/src/xcanfd.c#L1483

/**
* This Routine returns the Free Buffer out of 32 Transmit Buffers.
*
* @param	InstancePtr is a pointer to the XCanFd instance to be worked on.
*
*
* @return	Returns Free buffer if any free buffer
*		otherwise returns XST_NOBUFFER.
*
* @note		None.
*
******************************************************************************/
u32 XCanFd_GetFreeBuffer(XCanFd *InstancePtr)
{

	u32 RegVal;
	u32 Index = 0;

	Xil_AssertNonvoid(InstancePtr != NULL);
	Xil_AssertNonvoid(InstancePtr->IsReady == XIL_COMPONENT_IS_READY);

	RegVal = XCanFd_ReadReg(InstancePtr->CanFdConfig.BaseAddress, XCANFD_TRR_OFFSET);
	while((RegVal & ((u32)1 << Index)) != (u32)0) {
		Index++;
		if(Index == MAX_BUFFER_INDEX) {
			break;
		}

	}
	if(Index == MAX_BUFFER_INDEX) {
		return XST_NOBUFFER;
	}

	return RegVal;
}

Please check the below link to the usage of return value in the driver
https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/canfd/src/xcanfd.c#L484

sneeli-git, 2022-04-01:
Hi, 
Understood your concern and working on it to fix the problem.