Xilinx/embeddedsw #191: Unclear comment or missing check in function XEmacPs_BdRingFromHwRx

kk-chesse, 2022-03-23:
While searching for a memory leak in situations with low buffer counts I stumbled upon an integer underflow in the receive BdRing HwCnt. The underflow appears in the XEmacPs_BdRingFromHwRx function.

```c
unsigned XEmacPs_BdRingFromHwRx(XEmacPs_BdRing * RingPtr, unsigned BdLimit,
				 XEmacPs_Bd ** BdSetPtr)
{
	XEmacPs_Bd *CurBdPtr;
	u32 BdStr = 0;
	unsigned BdCount;
	unsigned BdPartialCount;

	CurBdPtr = RingPtr->HwHead;
	BdCount = 0;
	BdPartialCount = 0;
 
	/* If no BDs in work group, then there's nothing to search */
	if (RingPtr->HwCnt == 0) {
		*BdSetPtr = NULL;
		return (0);
	}
 
	/* Starting at HwHead, keep moving forward in the list until:
	 *  - A BD is encountered with its new/used bit set which means
	 *    hardware has completed processing of that BD.
	 *  - RingPtr->HwTail is reached and RingPtr->HwCnt is reached.
	 *  - The number of requested BDs has been processed
	 */
	while (BdCount < BdLimit) {
 
		/* Read the status */
		BdStr = XEmacPs_BdRead(CurBdPtr, XEMACPS_BD_STAT_OFFSET);
 
		if (!(XEmacPs_BdIsRxNew(CurBdPtr))) {
			break;
		}
 
		BdCount++;
 
		/* hardware has processed this BD so check the "last" bit. If
                 * it is clear, then there are more BDs for the current packet.
                 * Keep a count of these partial packet BDs.
		 */
		if (BdStr & XEMACPS_RXBUF_EOF_MASK) {
			BdPartialCount = 0;
		}
		else {
			BdPartialCount++;
		}
 
		/* Move on to next BD in work group */
		CurBdPtr = XEmacPs_BdRingNext(RingPtr, CurBdPtr);
	}
 
	/* Subtract off any partial packet BDs found */
	BdCount -= BdPartialCount;
 
	/* If BdCount is non-zero then BDs were found to return. Set return
	 * parameters, update pointers and counters, return success
	 */
	if (BdCount > 0) {
		*BdSetPtr = RingPtr->HwHead;
		RingPtr->HwCnt -= BdCount;
		RingPtr->PostCnt += BdCount;
		XEMACPS_RING_SEEKAHEAD(RingPtr, RingPtr->HwHead, BdCount);
		return (BdCount);
	}
	else {
		*BdSetPtr = NULL;
		return (0);
	}
}
```

The unclear part for me is in the comment above the while loop:

`*  - RingPtr->HwTail is reached and RingPtr->HwCnt is reached.`

Nowhere in the while loop are the variables RingPtr->HwTail or RingPtr->HwCnt checked

This can lead to BdCount = BdLimit and BdCount > RingPtr->HwCnt, what in turn leads to an underflow in RingPtr->HwCnt.

So where I'm left wondering is: is the RingPtr->HwTail andRingPtr->HwCnt check missing or is the comment wong?

colin-foster-in-advantage, 2022-03-30:
I found and reported this several years ago. I received no response, despite several others independently recreating the issue.

https://support.xilinx.com/s/question/0D52E00006hpmNcSAI/ip-rx-stack-crashes?language=en_US

Maybe you'll have more luck than I in getting this fix supported - especially since I'm pretty sure partial BDs is broken in my fix, but we don't use them.

HariniKatakamX, 2022-03-31:
Thanks, we're looking into this bug and it will be fixed in the second release of 2022.

HariniKatakamX, 2023-04-24:
This was fixed in the 2022.2 https://github.com/Xilinx/embeddedsw/commit/d8b78f75e4ffc5421086fed064afebea58cdf196

kk-chesse, 2023-04-24:
Thank you for your fix

closed 2023-04-24 by kk-chesse