Xilinx/embeddedsw #198: xrfclk.c reading registers before writing LMK04828 register 0x15F

OliverGorman, 2022-04-26:
file: https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/board_common/src/rfclk/src/xrfclk.c

## Issue
Reading register values back from the LMK04828 over SPI requires its register `0x15F` to be set to a value similar to `3E` as per the datasheet. However, calling `XRFClk_ReadReg` will cause whatever is stored in `MuxOutRegStorage` to be written to this register. This is problematic since the value of `MuxOutRegStorage` is uninitialised before `XRFClk_WriteReg` is called on register `0x15F`. The result is that if register `0x15F` has not been written to in the lifetime of the program but still has the correct value on-chip, reading back from the chip is silently prevented since the register is immediately overwritten with the incorrect value by `XRFClk_ReadReg`.

## Workaround
Write the appropriate value to `0x15F` using `XRFClk_WriteReg` before attempting to read any other registers.

## Proposed fix
Attempting to read a register while `MuxOutRegStorage` should at least raise an error and inform the user that they first need to set register `0x15F` to define how to read back.

xlnx-dcvetic, 2022-04-26:
Well spotted and I agree with the solution, but I would add that the proposed solution is the only solution.
The reason for this statement is that the register 0x15f is multi-functional and setting it in to read mode, even for the short period would break a previously set functionality.

Enhancement will include a change in XRFClk_ReadReg API only:
- read register 0x15f before reading a register from lmk04828,
- check the state of the register
- LOG the warning message

OliverGorman, 2022-04-28:
I think the main issue is that `MuxOutRegStorage` is undefined before the first write, so any attempt to use it before it has been set should be an error. Perhaps it should just be initialised to some invalid value so that this check can take place?