Xilinx/embeddedsw #200: Using LWIP with the Cortex-R5 doesn't seem to be possible

k7g03z, 2022-05-06:
I am trying to get Ethernet up and running with the ZCU104 Dev board with the Cortex-R5 core, not the Cortex-A53 core.

I set up a everything with the application LWIP UDP Performance Server on the R5 core as a standalone system and I get ethernet link, but only DMA errors when DHCP tries to acquire the IP address. After digging deeper in the code, it looks like the define flags only allow arm with aarch64 (for the Cortex-A53 core), see for example: 

https://github.com/Xilinx/embeddedsw/blob/b9b64f53e11723c8df0dfda1c59742428b6f1df1/ThirdParty/sw_services/lwip211/src/contrib/ports/xilinx/netif/xadapter.c#L242

I verified that everything works just fine when I do the same exact process but target the A53 core...

Has anyone gotten LWIP and ethernet to work with the Cortex-R5 core?

HariniKatakamX, 2022-05-06:
Hi,

The referenced code is run for ARMR5 as well (on account of __arm__). Could you please let us know what error messages you are getting?
Is the application compiled with an associated bsp for UDP server application (or is an existing bsp associated? this could change the lwip config parameters).

k7g03z, 2022-05-09:
I have retried the entire setup from scratch, this time creating a new `.xsa`  based on the git repo hardware description of the `ZCU104 Evaluation Board`. I don't know how my first `.xsa` is different than the new `.xsa`, but it now seems to work as expected. 

I now realize what I wrote regarding the  `#if defined` is incorrect. That should work for `arm32` and `arm64` just fine. 

I previously was getting errors from the LWIP stack from `xemacpsif_hw.c`:
`Receive DMA error`,  `Transmit DMA error`

If you have any suggestion what to look for in the old design, please let me know. Thanks!

closed 2022-06-30 by k7g03z