Xilinx/embeddedsw #202: Mapping between safety lounge PMU FW version (1.0, 1.2, 1.3 etc) to commits in this repo

sr-phm, 2022-05-12:
In the safety lounge there is zip filed archives of this repo and labeled version 1.0 and 1.2 etc.
I do not find howto map those versions to a specific commit here in this repo.
For example, if we use commit 26c14d9861010a0e3a55c73fb79efdb816eb42ca, what would that be? a pre-1.0 variant?

sarascha, 2022-05-16:
Prior to 2020.2, the public release and safety lounge releases differ slightly in commit history. 
2020.2 public release corresponds to 2020.2 v1.0 Safe PMU FW source code, drivers & BSP release on safety lounge. 
From 2020.2 onwards, the public releases are same as safety lounge releases, and these versions will be marked with v1.0.
In some cases, if there are changes to the public release, we release v1.1 and so on.

sr-phm, 2022-05-16:
Thanks for answer. I'm not fully clear yet though, I see there is source differences also.
Today we run 2019.1 : 
phm@gollum_apu:/usr/src/app/build/104298/build/tmp/work/plnx_zynqmp-xilinx-linux/pmu-firmware/2019.1+gitAUTOINC+26phm@gollum_apu:/usr/src/app/build/104298/build/tmp/work/plnx_zynqmp-xilinx-linux/pmu-firmware/2019.1+gitAUTOINC+26c14d9861-r0/git$ git status
On branch release-2019.1
Your branch is up-to-date with 'origin/release-2019.1'.
nothing to commit, working directory clean
phm@gollum_apu:/usr/src/app/build/104298/build/tmp/work/plnx_zynqmp-xilinx-linux/pmu-firmware/2019.1+gitAUTOINC+26c14d9861-r0/git$ ls
doc  lib  license.txt  mcap  README.txt  ThirdParty  XilinxProcessorIPLib
phm@gollum_apu:/usr/src/app/build/104298/build/tmp/work/plnx_zynqmp-xilinx-linux/pmu-firmware/2019.1+gitAUTOINC+26c14d9861-r0/git$ 

And we want to run safety lounge 1.3 variant : 
phm@bro-lt-3jn2sq2:~/sr/projects/pmu-fw-upgrade/1.3/embeddedsw_release_2019.1_web_v1.3$ ls embeddedsw/
lib  XilinxProcessorIPLib
phm@bro-lt-3jn2sq2:~/sr/projects/pmu-fw-upgrade/1.3/embeddedsw_release_2019.1_web_v1.3$ 

As seen, there seems to be code differences (more than slightly :) ) as well.
So you are saying the public release and safety release are forks?
I've tried to build the meta-xilinx pmu-firmware with a bbappend to update to the safety 1.3 variant but that fails due to missing part in the safety release. So we should then probably look at a pmu-firmware-safety recipe instead.
It seems to me something that more customers would do, so is there already such a recipe?
(I understand the safety variant is only guaranteed when using the prebuilt binary, but here we want to build exact same thing from source instead for other reasons)

sarascha, 2022-05-16:
Hi sr-phm,
for 2019.1: public release and safety lounge release (2019.1_web 1.3) are different. 
We normally compile pmu fw source using xilinx sdk/vitis tools to generate binary. We assume petalinux uses public released pmu fw source for binary generation.  We hope petalinux user guide UG1144 has more details on pmu fw source replacement and build process. pmu fw binary posted on safety lounge was generated using xilinx sdk/vitis tool chain.

sr-phm, 2022-05-16:
Yes, this gets clear now and the need for a pmu-firmware-safety recipe is clear. I was just wondering why such didn't already exist. Don't find any pointers in the doc you refer to, so I guess we just have to make this ourself then.
Thanks anyway for clearing out that these are really two different things, public vs safety pmu.
In future it would be nice if the safety was built from a branch of the embeddedsw repo similar to the current public one, so that if a customer wants to build the safety pmu it would be to change the recipe via a simple bbappend or so.

sarascha, 2022-05-17:
yes in future the safety pmu fw (lounge release) would be same as the public one. This has already started from 2020.2 release onwards.