Xilinx/embeddedsw #203: XSdPs_CmdTransfer() times out when trying to go from eMMC tran to stby state

dnygren, 2022-05-14:
The sdps driver has a XSdPs_Select_Card() function to go from the stby to tran state using CMD7 and an eMMC card's Relative Card Address (RCA).
https://github.com/Xilinx/embeddedsw/blob/b9b64f53e11723c8df0dfda1c59742428b6f1df1/XilinxProcessorIPLib/drivers/sdps/src/xsdps_options.c#L694

It should be possible to go back to the stby state from the trans state using a RCA of 0x0000 with a similar function that deselects the card like:

```
s32 XSdPs_Deselect_Card (XSdPs *InstancePtr)
{
    s32 Status;

    Xil_AssertNonvoid(InstancePtr != NULL);
    Xil_AssertNonvoid(InstancePtr->IsReady == XIL_COMPONENT_IS_READY);

    /* Send CMD7 - Deselect card with RCA 0x0000*/
    Status = XSdPs_CmdTransfer(InstancePtr, CMD7, x0000U, 0U);

    return Status;
}
```
[image]

It is necessary to do this in order to be able to re-read the CSD register with another XSdPs_GetCsd() call https://github.com/Xilinx/embeddedsw/blob/b9b64f53e11723c8df0dfda1c59742428b6f1df1/XilinxProcessorIPLib/drivers/sdps/src/xsdps_card.c#L622 that issues a CMD9 to check if a write to the CSD register was successful.

I added the function XSdPs_Deselect_Card() to the sdps driver but was unable to go from tran -> stby .

So it appears I get from stby -> tran fine using XSdPs_Select_Card(), (I check I'm in tran with a CMD13), but going from tran -> stby with the  XSdPs_Deselect_Card() function I made by just changing the RCA to 0x0000 using XSdPs_Select_Card() as a template experiences a XSdPs_CmdTransfer() time out :

```
.
.
.
*** In XSdPs_CmdTransfer() with Cmd = 3 ***                                 
****** CmdTransfer Normal Interrupt Status Reg = 0x0                        
****** CmdTransfer Normal Interrupt Status Reg = 0x1
!!!! CMD3 PASSED !!!!                       
*** In XSdPs_CmdTransfer() with Cmd = 7 & RCA ***
!!!! CMD7 -> tran PASSED !!!!                                                           
RESP REG0 0x700  <- was in stby state                                                                
RESP REG1 0x500                                                                 
RESP REG2 0x0                                                                   
RESP REG3 0x0                                                                   
    *** In XSdPs_CmdTransfer() with Cmd = 13 ***                                
    ****** CmdTransfer Normal Interrupt Status Reg = 0x0                        
    ****** CmdTransfer Normal Interrupt Status Reg = 0x1                        
!!!! CMD13 PASSED !!!!                                                          
RESP REG0 0x900 <- was in tran state                                                                
RESP REG1 0x0                                                                   
RESP REG2 0x0                                                                   
RESP REG3 0x0                                                                   
    *** In XSdPs_CmdTransfer() with Cmd = 7 & RCA 0x0000 ***                                 
    ****** CmdTransfer Normal Interrupt Status Reg = 0x0                        
    ****** CmdTransfer Normal Interrupt Status Reg = 0x8000                     
    ****** CmdTransfer Error Interrupt Status Reg = 0x1                         
    ****** CmdTransfer Present State Lower Status Reg = 0x1                     
    ****** CmdTransfer Present State Upper Status Reg = 0x1FF                   
    ****** CmdTransfer Error Interrupt Status Reg = 0x1                         
!!!! CMD7 -> stby Failed !!!!
```

saikrishnapotthuri, 2022-05-25:
Hi,
Card is not expected to send any response while deselecting the card. Default CMD7 defined in the SDPS driver is expecting R1 response as this is meant to be developed for card selection.
https://github.com/Xilinx/embeddedsw/blob/b9b64f53e11723c8df0dfda1c59742428b6f1df1/XilinxProcessorIPLib/drivers/sdps/src/xsdps_host.c#L1661
While implementing card deselect, we need to make sure that no response is expected from the Card (even cross checked the Linux implementation, not expecting any response during card deselect), so define a new command for deselection or manage the CMD7 implementation for card deselect for not to include R1 response in the XSdPs_FrameCmd() API.
Tested with this change at my end, Card deselect is working fine.

dnygren, 2022-05-25:
Can you share your card deselect code? If not can you point me at the Linux implementation? I looked through the mmc-utils https://git.kernel.org/pub/scm/utils/mmc/mmc-utils.git but wasn't able to determine how they did deselect.

saikrishnapotthuri, 2022-05-26:
Major difference between Select and Deselect the card is Card address and Response type.
Card Select-> RCA + R1 response
Card DeSelect-> 0x0 + No response

In SDPS driver defined below dummy command and code changes for testing purpose
#define MCMD70	 0x7700U

In XSdPs_FrameCmd() API, handle MCMD70 command in the following way
case MCMD70:
		Cmd = CMD7;
		RetVal = Cmd;
		break;

s32 XSdPs_DeSelect_Card (XSdPs *InstancePtr)
{
	s32 Status;

	Xil_AssertNonvoid(InstancePtr != NULL);
	Xil_AssertNonvoid(InstancePtr->IsReady == XIL_COMPONENT_IS_READY);

	Status = XSdPs_CmdTransfer(InstancePtr, MCMD70, 0x0, 0U);
	return Status;
}

Linux implementation
https://github.com/Xilinx/linux-xlnx/blob/master/drivers/mmc/core/mmc_ops.c#L91

dnygren, 2022-05-26:
Thank you for sharing that! I would have never figured out there was no response because the eMMC spec I am working from is JESD84-B50 which says on page 111 Table 41 that CMD7 has a response of either R1 or R1b :

[image]

[image]

 I don't see anywhere in the spec where it is stated that there is no response when going from tran to stby. In "Note 1" they do not discuss this type of state transition either.

closed 2022-05-26 by dnygren