Xilinx/embeddedsw #204: some questions about IIS driver

kangliyu1, 2022-05-19:
Hello, when I compile the IIC driver, there are some problems as shown in the figure below. I suspect that it is caused by the incompleteness or error of the makefile, and there are no lib and include folders in the makefile in the folder downloaded from github. I don't know where these two folders can be found, I hope someone can give me some pointers, thank you very much
[image]

gcnugoud, 2022-05-19:
hi,

Can we know which tool your using for compiatio  vitis/xsct or your own flow.
lib and include folders created by vitis/xsct tools during build.
Makefiles are developed to be aligned with vitis/xsct xilinx environment.

Thanks,
Srinivas

kangliyu1, 2022-05-20:
> hi,
> 
> Can we know which tool your using for compiatio vitis/xsct or your own flow. lib and include folders created by vitis/xsct tools during build. Makefiles are developed to be aligned with vitis/xsct xilinx environment.
> 
> Thanks, Srinivas

Hello, I installed the arm compiler myself in the linux environment and compiled it directly, but the lib folder was not generated after compilation, and the error in the above picture was reported.

[image]

mubinsyed, 2022-05-23:
Hi @kangliyu1 ,

Can you please provide following details,

- For which processor you are trying to compile the code
- are you using Vitis/xsct for Building BSP/applications?. As driver makefiles are aligned with Vitis/xsct infrastructure, you need 
 to use Vitis/xsct to create the BSP/application project

kangliyu1, 2022-05-25:
> * For which processor you are trying to compile the code
> * are you using Vitis/xsct for Building BSP/applications?. As driver makefiles are aligned with Vitis/xsct infrastructure, you need
>   to use Vitis/xsct to create the BSP/application project

@mubinsyed 
Hello, I am missing some related files when compiling, where should I find these files?
[image]