Xilinx/embeddedsw #206: Snap shot the latest BD when address space is more than 32 bit in axidma driver

ToTamire, 2022-05-19:
Function XAxiDma_BdRingSnapShotCurrBd  inside file xaxidma_bdring.h don't copy upper 32 bits for Cur BD REG when address space is more than 32 bits.
```c
#define XAxiDma_BdRingSnapShotCurrBd(RingPtr)		  \
	{								  \
		if (!RingPtr->IsRxChannel) {				  \
			(RingPtr)->BdaRestart = 			  \
				(XAxiDma_Bd *)(UINTPTR)XAxiDma_ReadReg(		  \
					(RingPtr)->ChanBase,  		  \
					XAXIDMA_CDESC_OFFSET);		  \  // only XAXIDMA_CDESC_OFFSET without XAXIDMA_CDESC_MSB_OFFSET
		} else {						  \
			if (!RingPtr->RingIndex) {				  \
				(RingPtr)->BdaRestart = 		  \
				(XAxiDma_Bd *)(UINTPTR)XAxiDma_ReadReg(            \
					(RingPtr)->ChanBase, 		  \
					XAXIDMA_CDESC_OFFSET);		  \  // only XAXIDMA_CDESC_OFFSET without XAXIDMA_CDESC_MSB_OFFSET
			} else {					  \
				(RingPtr)->BdaRestart = 		  \
				(XAxiDma_Bd *)(UINTPTR)XAxiDma_ReadReg( 		  \
				(RingPtr)->ChanBase,                      \
				(XAXIDMA_RX_CDESC0_OFFSET +		  \  // only XAXIDMA_RX_CDESC0_OFFSET without XAXIDMA_RX_CDESC0_MSB_OFFSET
                                (RingPtr->RingIndex - 1) * 		  \
					XAXIDMA_RX_NDESC_OFFSET)); 	  \
			}						  \
		}							  \
	}
```

ToTamire, 2022-05-19:
Probably same issue with function XAxiDma_BdRingGetCurrBd in same file.

HariniKatakamX, 2022-05-20:
Thanks, will provide fix in both functions in the next release