Xilinx/embeddedsw #209: some questions about  driver compile

kangliyu1, 2022-05-31:
Hello! When I downloaded this open source project and compiled it in vitis, the relevant files were missing, where should I find these files? (Take the I2C driver as an example, I made an error during the compilation process, I Going to the upper layers of this directory to find the relevant files and adding them to the path, there is still an error, so I would like to ask where the missing files should be found.)

By the way, how can I use vitis to compile the makefile of this project?

[image]

NghiSpectrohm, 2022-05-31:
It is likely that you need to add library "metal" to your compiler linkage in vitis; if makefile from command line you need to add -lmetal.  From a Xilinx reference design example:
[image]
[image]

xil_ libraries are xilinx metal that you can find reference design from their board support reference packages.

kangliyu1, 2022-06-09:
> It is likely that you need to add library "metal" to your compiler linkage in vitis; if makefile from command line you need to add -lmetal. From a Xilinx reference design example: [image] [image] [...]

@NghiSpectrohm Hello, after I added it according to your method, the error as shown in the figure appeared after building the project. The constitution may ask you, how can I solve this problem?
[image]

NghiSpectrohm, 2022-06-09:
XPAR definition is defined within "xparameters.h" that either comes with your reference board or your unique Vivado hardware platform.  

Did you follow the reference design by Xilinx for IIC libraries?  It is found under: 

`/Vitis/<version>/data/embeddedsw/XilinxProcessorIPLib/drivers/iic_v3_9/examples/`

Add this to your main.h:

`#include "xparameters.h" `

kangliyu1, 2022-06-09:
> #include "xparameters.h"

@NghiSpectrohm 
Thank you for your reply, in the above error, I just ran the built-in example, but still reported this error

[image]

NghiSpectrohm, 2022-06-09:
This is my CLK104 application in Vitis that used Xilinx library:
`/*
 * The following constants map to the XPAR parameters created in the
 * xparameters.h file. They are defined here such that a user can easily
 * change all the needed parameters in one place.
 */
#ifdef __BAREMETAL__
#define RFDC_DEVICE_ID 	XPAR_XRFDC_0_DEVICE_ID
#define I2CBUS	1
#define XRFDC_BASE_ADDR		XPAR_XRFDC_0_BASEADDR
#define RFDC_DEV_NAME    XPAR_XRFDC_0_DEV_NAME
#else
#define RFDC_DEVICE_ID 	0
#define I2CBUS	12
#endif`

Likely you are not using BAREMETAL for your development, or your hardware platform has not set up properly.  As you see in the above reference code, Metal library invokes xparameter.h.

Try to clean both your hardware platform and application, then rebuild platform, then your codes again.  Sometimes Vitis got stuck with temp files and do a clean build would clear some errors.

Lastly, try to use Xilinx IIC reference design in /example folder first, then adding more codes as required.  It would help to get one basic version by Xilinx reference to run.

kangliyu1, 2022-06-09:
> This is my CLK104 application in Vitis that used Xilinx library: `/*
>
> * The following constants map to the XPAR parameters created in the
> * xparameters.h file. They are defined here such that a user [...]

Still an error was reported, is there an error in the handshake step?

[image]
[image]
[image]
[image]

NghiSpectrohm, 2022-06-09:
That is no, you picked the wrong properties.  You have to right click on "application", not "system" or "source file".  See highlight here for my project.  
[image]

In my screenshot, Application is af-clk that is part of af-clk system with source file as show.

Also, download Xilinx reference design for ZCU104 would also have a powerpoint presentation how to build and create Vitis app for hardware application.  It has screenshots step by step.

kangliyu1, 2022-06-10:
> Xilinx reference design for ZCU104
Hello, at the beginning this library was blank, after I added matel, it still reported an error
I found ug1400, referred to it, and found that there is no compilation tutorial that explains the driver
Please how can I Link to Baremetal IIC libraries:

[image]

[image]

anirudha1977, 2022-06-10:
It looks like the IIC driver is not being pulled in. Have you checked the libsrc folder to see which all drivers are pulled into the project?

regards

kangliyu1, 2022-06-10:
> libsrc
Do I need to add the files in the src of iic to the directory of src? As shown below.
[image]

anirudha1977, 2022-06-10:
libsrc is the folder that vitis generates by pulling in all drivers for HW blocks that are present in a design.
As an example, if I create my vitis project at a folder with name "vitisp" and it is for zc702 (Zynq) board, the libsrc path will be at:
....../vitisp/zc702/ps7_cortexa9_0/standalone_ps7_cortexa9_0/bsp/ps7_cortexa9_0/libsrc

For iic (if your design has iic) you will have a folder with name "iic_v3*".  If it is not there (which I fear would be the case), we need to check the HW design itself.
The framework pulls the driver automatically if it sees the relevant HW IP/block.

regards

kangliyu1, 2022-06-10:
> libsrc is the folder that vitis generates by pulling in all drivers for HW blocks that are present in a design. As an example, if I create my vitis project at a folder with name "vitisp" and it is [...]
@anirudha1977 thank you for your reply ！ I ran the example in the figure below. Although there is this directory in libsrc, an error is still reported.
[image]

[image]

[image]

anirudha1977, 2022-06-10:
Your error clearly says that libmetal is not included in the project for some reason.
In your folder list, I dont see libmetal as well.

So, in the BSP settings, you need to include libmetal. Once you regenerate the BSP, in the same libsrc folder, you can see libmetal pulled in.
Then you should not see this issue.

kangliyu1, 2022-06-10:
> Your error clearly says that libmetal is not included in the project for some reason. In your folder list, I dont see libmetal as well.
>
> So, in the BSP settings, you need to include libmetal. Once [...]

@anirudha1977 

How should I add this libmetal?
[image]
[image]

anirudha1977, 2022-06-10:
To add, you just need to select libmetal checkbox and that's it. You dont have to worry to get into tool settings and do anything.
If you dont see libmetal folder, that is strange.
May be you can start afresh and see if you still see the issue, then it would be really strange.

kangliyu1, 2022-06-10:
> To add, you just need to select libmetal checkbox and that's it. You dont have to worry to get into tool settings and do anything.
> If you dont see libmetal folder, that is strange.
> May be you can start afresh and see if you still see the issue, then it would be really strange.
@anirudha1977 thank you for your reply! I also want to ask, if it is not in libsrc, how to add it?

anirudha1977, 2022-06-10:
libsrc is tool generated. Tool can only add it. And tool adds it based on the checkbox. 
After you select checkbox against libmetal, regenerate BSP sources. Once that is done, please check libsrc. If by then libmetal does not show up, then there is a bug somewhere.

kangliyu1, 2022-06-10:
> libsrc is tool generated. Tool can only add it. And tool adds it based on the checkbox.
> After you select checkbox against libmetal, regenerate BSP sources. Once that is done, please check libsrc. If by then libmetal does not show up, then there is a bug somewhere.

> libsrc is tool generated. Tool can only add it. And tool adds it based on the checkbox. After you select checkbox against libmetal, regenerate BSP sources. Once that is done, please check libsrc. If by then libmetal does not show up, then there is a bug somewhere.

@anirudha1977 So, how can the relevant drivers in libsrc be ported to the processor? For example, how can I port the relevant driver to a certain RISCV processor for use

anirudha1977, 2022-06-10:
Drivers are attached to a HW block/IP and are independent of processor. The Xilinx drivers do not have any property that associates them with a processor. 

If you have installed Vitis, you can typically look for any drivers folder at
...<Vitis project>/installs/lin64/Vitis/2021.1/data/embeddedsw/XilinxProcessorIPLib/drivers

Inside the relevant driver folder, you can see a <driver_name>.mdd file. Vitis tool reads this file.  The _supported_peripherals attribute in this file is used by Vitis to pull in drivers.

Libraries (like libmetal) however have processor attachment.
Like drivers, libraries have a <library_name>.mld file that the Vitis tool refers. For libmetal, you can see the original source at:

...<Vitis_Install_Dir>/installs/lin64/Vitis/2021.1/data/embeddedsw/ThirdParty/sw_services
Pick the latest libmetal version, go into the data folder and look for the *.mld file.
libmetal does not have a supported_peripherals attribute. Had it been there, it would have meant that libmetal is being supported for a certain set of processors. Since it is not there, libmetal is supported for all processors.

regards,

kangliyu1, 2022-06-13:
> Drivers are attached to a HW block/IP and are independent of processor. The Xilinx drivers do not have any property that associates them with a processor.
> 
> If you have installed Vitis, you can typically look for any drivers folder at .../installs/lin64/Vitis/2021.1/data/embeddedsw/XilinxProcessorIPLib/drivers
> 
> Inside the relevant driver folder, you can see a <driver_name>.mdd file. Vitis tool reads this file. The _supported_peripherals attribute in this file is used by Vitis to pull in drivers.
> 
> Libraries (like libmetal) however have processor attachment. Like drivers, libraries have a <library_name>.mld file that the Vitis tool refers. For libmetal, you can see the original source at:
> 
> ...<Vitis_Install_Dir>/installs/lin64/Vitis/2021.1/data/embeddedsw/ThirdParty/sw_services Pick the latest libmetal version, go into the data folder and look for the *.mld file. libmetal does not have a supported_peripherals attribute. Had it been there, it would have meant that libmetal is being supported for a certain set of processors. Since it is not there, libmetal is supported for all processors.
> 
> regards,
@anirudha1977 thank you for your reply！
Hello, I imported the iic example of this github into the IDE for compilation. Even if there is a libmatel folder, the error of missing files is still reported after the build. Does the IIC of this github do not match the libmatel of the IDE platform?

[image]
[image]
[image]

kangliyu1, 2022-06-14:
> @anirudha1977 your explanation above is very helpful! I learned new things.
> 
> @kangliyu1 look like you still have trouble compiling your VITIS application. Per @anirudha1977: `Drivers are attached to a HW block/IP`
> 
> What reference hardware platform in VIVADO that you exported your platform for VITIS? Did you have an instance for I2C IP in your hardware code? Can you screenshot your design in VIVADO? Perhaps you did not have I2C created in your VIVADO design and VITIS does not pulled IIC library as your hardware does not needed I2C drivers.
@NghiSpectrohm Hello, I did not export my platform for VITIS in vivado, is this necessary? Because if you compile the example that comes with it, you don't need to add the content exported by vivado, and this error will not occur, but the compilation will pass. Can you give me some pointers?
[image]
[image]

NghiSpectrohm, 2022-06-14:
@kangliyu1 I commented you for trying to learn Xilinx SoC design.  However without basic background of designing and integrating a complex PL (programmable logic) and PS ( processing system) system on a chip (SoC) using Xilinx prototyping boards, I'm afraid you started in the wrong direction.

VIVADO: you can do everything in PL without a need of VITIS application for a PS.  This is most people starts for their hardware design.

VITIS: application development targeting PS running ARM, Linux, standlone RTOS, etc, but your OS (operating system) will need to interface with external hardware peripherals (like this I2C port example).  You should start writing a "hello world" equivalent in VITIS first using a pre-built BSP (board support package), then move to more advance topics like read write memory address.  

Your code failed because you randomly picked up a VITIS project and expected it to compile and run.  You should try the "hello world" examples first because it does not require external resources.  I'm afraid this is the last I try to help you.  **You should close this Github issue** as it is not Xilinx embedded issues with Xilinx BSP & drivers.  

There are many great resources for starting out VITIS projects.  Start with simple ones, then advance slowly.  Use Xilinx Support forum for your development issue (https://support.xilinx.com/); not Xilinx Github as this Git is for reporting bugs and features for embedded system and not your every day coding issues.