Xilinx/embeddedsw #214: using extern keyword within macro cause error when using c++ with namespaces

karaketir16, 2022-07-15:
in file [ThirdParty/bsp/freertos10_xilinx/src/Source/portable/GCC/ARM_CA9/portmacro.h](https://github.com/Xilinx/embeddedsw/blob/b3d8b420b421730ea505da55b42174dc90f885c1/ThirdParty/bsp/freertos10_xilinx/src/Source/portable/GCC/ARM_CA9/portmacro.h) and other ports

```
/* Called at the end of an ISR that can cause a context switch. */
#define portEND_SWITCHING_ISR( xSwitchRequired )\
{												\
extern uint32_t ulPortYieldRequired;			\
												\
	if( xSwitchRequired != pdFALSE )			\
	{											\
		ulPortYieldRequired = pdTRUE;			\
	}											\
}
```

when i use this macro inside a namespace, like 
```
namespace nms {
    void myFunction(){
        portEND_SWITCHING_ISR( pdTRUE );
    }
}
```

it causes an error
`undefined reference to 'nms::ulPortYieldRequired'`

to fix this, I suggest, move that line outside the macro, so that every place reaching the macro, can reach the variable. 

```
extern uint32_t ulPortYieldRequired;			
/* Called at the end of an ISR that can cause a context switch. */
#define portEND_SWITCHING_ISR( xSwitchRequired )\
{												\
												\
	if( xSwitchRequired != pdFALSE )			\
	{											\
		ulPortYieldRequired = pdTRUE;			\
	}											\
}
```

anirudha1977, 2022-07-18:
Thanks @karaketir16 
Valid point. We will get this fixed in next release.

karaketir16, 2024-06-15:
Hi,

I noticed that the issue was solved by commit 9cf6a89, which was based on the patch I submitted. Could you please add me to the contributors list on GitHub? I would appreciate it.

Thank you!

closed 2024-07-05 by karaketir16