Xilinx/embeddedsw #218: PMU Firmware build fails when ENABLE_WDT is defined on Vivado2022.1

phlmm, 2022-09-02:
I'm seeing the folowing build errors when I try to rebuild pmu firmware (either from xsct, manually or yocto).

``` 
mb-gcc  -mlittle-endian -mxl-barrel-shift -mxl-pattern-compare -mxl-soft-mul -mcpu=v9.2 -MMD -MP      -mlittle-endian -mxl-barrel-shift -mxl-pattern-compare -mcpu=v9.2 -mxl-soft-mul -Os -flto -ffat-lto-objects -DENABLE_SCHEDULER -DENABLE_EM -DENABLE_PM -DENABLE_RECOVERY -DENABLE_ESCALATION -DENABLE_WDT -DCHECK_HEALTHY_BOOT -DENABLE_EFUSE_ACCESS -DEFUSE_ACCESS -DENABLE_FPGA_LOAD=0                     -c xpfw_mod_wdt.c -o xpfw_mod_wdt.o -Izynqmp_pmufw_bsp/psu_pmu_0/include -I.
xpfw_mod_wdt.c:20:10: error: #error "ENABLE_WDT is defined but psu_csu_wdt is not defined in the design"
   20 |         #error "ENABLE_WDT is defined but psu_csu_wdt is not defined in the design"
      |          ^~~~~
xpfw_mod_wdt.c: In function 'XPfw_WdtSetVal':
xpfw_mod_wdt.c:39:33: error: 'XPMU_PMUWDT_WDT_CLK' undeclared (first use in this function); did you mean 'XPMU_FPDWDT_WDT_CLK'?
   39 | #define XPFW_WDT_CLK_PER_MSEC ((XPMU_PMUWDT_WDT_CLK) / (XPFW_WDT_PRESCALER * 1000U))
      |                                 ^~~~~~~~~~~~~~~~~~~
xpfw_mod_wdt.c:84:49: note: in expansion of macro 'XPFW_WDT_CLK_PER_MSEC'
   84 |                 CounterValue = ((TimeOutVal) * (XPFW_WDT_CLK_PER_MSEC)) >> XPFW_WDT_CRV_SHIFT;
      |                                                 ^~~~~~~~~~~~~~~~~~~~~
xpfw_mod_wdt.c:39:33: note: each undeclared identifier is reported only once for each function it appears in
   39 | #define XPFW_WDT_CLK_PER_MSEC ((XPMU_PMUWDT_WDT_CLK) / (XPFW_WDT_PRESCALER * 1000U))
      |                                 ^~~~~~~~~~~~~~~~~~~
xpfw_mod_wdt.c:84:49: note: in expansion of macro 'XPFW_WDT_CLK_PER_MSEC'
   84 |                 CounterValue = ((TimeOutVal) * (XPFW_WDT_CLK_PER_MSEC)) >> XPFW_WDT_CRV_SHIFT;
      |                                                 ^~~~~~~~~~~~~~~~~~~~~
xpfw_mod_wdt.c: In function 'InitCsuPmuWdt':
xpfw_mod_wdt.c:120:44: error: 'XPMU_PMUWDT' undeclared (first use in this function); did you mean 'XPMU_LPDWDT'?
  120 |         WdtConfigPtr = XWdtPs_LookupConfig(XPMU_PMUWDT);
      |                                            ^~~~~~~~~~~
      |                                            XPMU_LPDWDT
xpfw_mod_wdt.c:39:33: error: 'XPMU_PMUWDT_WDT_CLK' undeclared (first use in this function); did you mean 'XPMU_FPDWDT_WDT_CLK'?
   39 | #define XPFW_WDT_CLK_PER_MSEC ((XPMU_PMUWDT_WDT_CLK) / (XPFW_WDT_PRESCALER * 1000U))
      |                                 ^~~~~~~~~~~~~~~~~~~
xpfw_mod_wdt.c:40:57: note: in expansion of macro 'XPFW_WDT_CLK_PER_MSEC'
   40 | #define XPFW_WDT_COUNTER_VAL ((XPFW_WDT_EXPIRE_TIME) * (XPFW_WDT_CLK_PER_MSEC))
      |                                                         ^~~~~~~~~~~~~~~~~~~~~
xpfw_mod_wdt.c:146:24: note: in expansion of macro 'XPFW_WDT_COUNTER_VAL'
  146 |         CounterValue = XPFW_WDT_COUNTER_VAL >> XPFW_WDT_CRV_SHIFT;
      |                        ^~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:35: xpfw_mod_wdt.o] Error 1
```

in my system.mss (generated from vivado2022.1) I have enabled two wdt modules of the PS.

```

BEGIN DRIVER
 PARAMETER DRIVER_NAME = ttcps
 PARAMETER DRIVER_VER = 3.15
 PARAMETER HW_INSTANCE = psu_ttc_0
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = ttcps
 PARAMETER DRIVER_VER = 3.15
 PARAMETER HW_INSTANCE = psu_ttc_1
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = ttcps
 PARAMETER DRIVER_VER = 3.15
 PARAMETER HW_INSTANCE = psu_ttc_2
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = ttcps
 PARAMETER DRIVER_VER = 3.15
 PARAMETER HW_INSTANCE = psu_ttc_3
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = uartps
 PARAMETER DRIVER_VER = 3.11
 PARAMETER HW_INSTANCE = psu_uart_1
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = generic
 PARAMETER DRIVER_VER = 3.1
 PARAMETER HW_INSTANCE = psu_usb_0
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = usbpsu
 PARAMETER DRIVER_VER = 1.11
 PARAMETER HW_INSTANCE = psu_usb_xhci_0
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = wdtps
 PARAMETER DRIVER_VER = 3.5
 PARAMETER HW_INSTANCE = psu_wdt_0
END

BEGIN DRIVER
 PARAMETER DRIVER_NAME = wdtps
 PARAMETER DRIVER_VER = 3.5
 PARAMETER HW_INSTANCE = psu_wdt_1
END
```

I'm using Vivado 2022.1, but the same error was visible on Vivado 2020.1 as well.

phlmm, 2022-09-19:
The system needs to be configured with CONFIG.PSU__PMU__PERIPHERAL__ENABLE {1}, otherwise the MSS file do not get the psu_csu_wdt:
BEGIN DRIVER
 PARAMETER DRIVER_NAME = wdtps
 PARAMETER DRIVER_VER = 3.5
 PARAMETER HW_INSTANCE = psu_csu_wdt
END

Coding-xinxiangwanji2017, 2022-09-19:
Hi，您好：    我收到了您邮件，等我看完邮件以后，第一时间给您回复。

closed 2022-09-19 by phlmm