Xilinx/embeddedsw #233: No FreeRTOS settings to override INCLUDE_xTaskGetCurrentTaskHandle

smartinou, 2022-12-09:
The freertos10_xilinx.tcl script that generates FreeRTOSConfig.h file doesn't allow to override the default setting for INCLUDE_xTaskGetCurrentTaskHandle, found in FreeRTOS.h. The default value is disabled.

Third party SafeFlash library from HCC requires this feature to be enabled.

Currently there's no way to get around this but to post-edit the FreeRTOSConfig.h file, which is far from being a good solution.

anirudha1977, 2022-12-10:
Thanks for pointing to this. We will take a look at this ask in our next release.

sivadur, 2023-02-21:
@smartinou Though INCLUDE_xTaskGetCurrentTaskHandle is not enabled by default, the xTaskGetCurrentTaskHandle() API should be available by default as we have configUSE_MUTEXES enabled by default.  https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/tasks.c#L4016

Also, it will be enabled by default in our next release with FreeRTOS upgrade as latest FreeRTOS kernel has this enabled by default https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#L277 and i hope that will fix your issue.