Xilinx/embeddedsw #237: xxxve Ethernet_ example_ intr_ Mcdma. c driver code problem

woshiybc, 2022-12-29:
1.Run xxxve Ethernet_ example_ intr_ Mcdma. c driver code problem
Can't jump out of while (! FramesTx);

anirudha1977, 2022-12-30:
Hi,
Can you please elaborate what driver code problem are you referring to?

regards,
Anirudha

woshiybc, 2022-12-30:
hello, Contains an example on how to use the XXxvetherent driver directly.

woshiybc, 2022-12-30:
Use supporting files xxxvethernet_example_util.c and xxxvethernet_example.h.
For details, see xxxvethernet_example_intr_mcdma.c.

HariniKatakamX, 2023-01-02:
Could you please elaborate if you are referring a specific driver code issue?
The line while (!FramesTx); will break when the frame transmission over XXV Ethernet IP has completed. Can you please check your design to make sure XXV and MCDMA IPs are setup properly. There is also an example design here - https://github.com/Xilinx-Wiki-Projects/ZCU102-Ethernet/tree/main/2019.2/pl_eth_10g

woshiybc, 2023-01-03:
Hello, I checked that the XXV and MCDMA IP settings are correct, because the IP addresses are the official XILINX IP addresses, and the driver is also an official routine. Why is the expected result of running the driver always waiting while (! FramesTx); Is there no recovery flag bit for hardware interrupt

woshiybc, 2023-01-03:
Another question is, what is the version of this code related to? For example, if I use Vitas and Vivado 2020.2, should the corresponding driver code versions be the same

HariniKatakamX, 2023-01-03:
> Another question is, what is the version of this code related to? For example, if I use Vitas and Vivado 2020.2, should the corresponding driver code versions be the same

Yes, that is recommended

HariniKatakamX, 2023-01-03:
> Hello, I checked that the XXV and MCDMA IP settings are correct, because the IP addresses are the official XILINX IP addresses, and the driver is also an official routine. Why is the expected result [...]

The wait loop will exit when the completion interrupt occurs (the completion interrupt increments FramesTx as per the expected behaviour)

woshiybc, 2023-01-03:
After the interruption is completed, the loop does not exit, and FramesTx does not increase automatically. Can I try to assign FramesTx manually? Is there any way to see the local loopback, and the data is successfully received and sent

woshiybc, 2023-01-04:
Hello, the platform we use is ZCU106. Is this code driver incompatible with ZCU106?

HariniKatakamX, 2023-04-24:
Yes, the driver is compatible with ZCU106. Can you please check your design and troubleshoot as per below debug tips?
Once the completion interrupt occurs, the received frames are processed (if there are no errors) and the count is increased. No, please do not update FramesTx manually. The example is set up to operate in loopback. You can also read the RX and TX statistics in MCDMA (valid and error packet counts are displayed there) to debug. You can also check the interrupt status register upon entry into the handler and dump the MCDMA buffer descriptors to check if packets are received.