Xilinx/embeddedsw #239: XFsbl_PollTimeout condition is only checked once

StefanoPietrosanti, 2023-01-13:
In xfsbl_misc.c line 896: ``if (cond) {``, cond is effectively a constant and does not evaluate any condition.
```
/*****************************************************************************/
/**
 * *
 * * This macro polls an address periodically until a condition is met or till the
 * * timeout occurs.
 * * The minimum timeout for calling this macro is 100us. If the timeout is less
 * * than 100us, it still waits for 100us. Also the unit for the timeout is 100us.
 * * If the timeout is not a multiple of 100us, it waits for a timeout of
 * * the next usec value which is a multiple of 100us.
 * *
 * * @param            Addr - Address to be polled
 * * @param            Value - variable to read the value
 * * @param            Cond - Condition to checked (usually involves VALUE)
 * * @param            TimeOutInUs - timeout in micro seconds
 * *
 * * @return           XFSBL_SUCCESS - when the condition is met
 * *                   XFSBL_FAILURE - when the condition is not met
 * * @note             none
 * *
 * *****************************************************************************/
s32 XFsbl_PollTimeout(u32 Addr,u32 Value, u32 cond, u32 TimeOutInUs)
{
	s32 Status;
	u64 timeout = TimeOutInUs / 100U;

	if ((TimeOutInUs % (100U)) != 0U) {
		timeout++;
	}

	for(;;) {
		Value = Xil_In32(Addr);
		if(cond) {
			break;
		} else {
			usleep(100U);
			timeout--;
			if(timeout==0U)
			break;
		}
	}

	if (timeout > 0U) {
		Status = XFSBL_SUCCESS;
	} else {
		Status = XFSBL_FAILURE;
	}

	return Status;
 }

```
The variable "cond" is not updated anywhere in the function, so the condition is not updated with the new read Value. 
This regression was introduced in commit 59f1dda4fd81d0a546c71dcbaaf4278d7b8b7a48 .

SaratCS, 2023-01-16:
In the calling function of XFsbl_PollTimeout(), a conditional statement’s result is passed to this third parameter (cond). Hence, if that condition is true in the calling function, cond will be 1, else will be 0. That is, the condition is already evaluated in the calling function and only its result will be checked in this function.
I don't see any issue in the code. However, I agree that the comments (3rd param description) could have described it more accurately.

StefanoPietrosanti, 2023-01-16:
If that was the case, this function will not make sense as it wouldn't have to re-read the value at each iteration until timeout.
This function is supposed to evaluate a condition each time the value is read and, while the condition doesn't evaluate to true, it will count down a timeout timer. The timeout timer is there to wait for the condition to eventually become true. 

If you go back to the commit that introduced this regression, you can see how this function replaces the following macro. where the condition was evaluated at each Xil_In32() read: 
```
#define Xil_poll_timeout(IO_func, ADDR, VALUE, COND, TIMEOUT_US) \
 ( {	  \
	u64 timeout = TIMEOUT_US/100;    \
	if(TIMEOUT_US%100!=0)	\
		timeout++;   \
	for(;;) { \
		VALUE = IO_func(ADDR); \
		if(COND) \
			break; \
		else {    \
			usleep(100);  \
			timeout--; \
			if(timeout==0) \
			break;  \
		}  \
	}    \
	(timeout>0) ? 0 : -1;  \
 }  )
```