Xilinx/embeddedsw #242: XSpi driver: Major loss of data in interrupt mode with TX HALF EMPTY flags

Wadosh42, 2023-01-24:
Bug injected in commit: a4acb9ecba683b6d7cb3d251bfb8c3ee832935c1

Hi,

During a project update, we have observe major drop of data on a master AXI QSPI IP (single lane) due to an incomplete fix in version 4.7 of XSpi driver in the interrupt handler. The previous version had a bug leading to an overrun of receive FIFO, but the fix is worse when you enable the TX HALF EMPTY as you can loss half of your data on each interrupt refilling iteration.

Previous refuelling was based on polling status bit until buffer is full or all data have been sent, this have been replaced by a for loop which only care about static fifo depth and remaining data. In case of TX HALF EMPTY, the interrupt will blindy attempt to insert a complete FIFO in a half empty FIFO...

Quick fix can be done just by setting a "DataLen" divided by two in case of TX HALF EMPTY, but I not a fan of blindy refuel and it still may have available slot in FIFO that we can take advantage of.

Reproduce:
- Configure an AXI IP QSPI as master in interrupt mode with TX HALF EMPTY interrupt unmasked
- Request a huge transfer exceeding 2 times FIFO depth
- Observe total transmitted bytes on bus (it will probably happen in loopback mode too if you don't have scope)

Amit-Kumar-Mahapatra, 2023-01-27:
Hello Wadosh42,

Thank you for catching this. We will fix it in next release, i.e., 2023.1

Regards,
Amit

Wadosh42, 2023-01-27:
Hi,

Thanks for the consideration! 😄 

Just another information, he have the RX OVERRUN bug with both my quick fix or the bug patched in version 4.8. I suggest to handle both RX and TX FIFO simultaneously with a single polling loop on status register instead of emptying RX first then filling TX.

Best regards

Amit-Kumar-Mahapatra, 2023-04-24:
Hello Wadosh42,

With curent flow, i.e., emptying RX first then filling TX are you seeing any failure ?