Xilinx/embeddedsw #247: freertos_intr_example.c example does not work on R5 core on ZCU104  - no interrupts are delievered

chris-laplante, 2023-03-08:
Please re-test the example found here: https://github.com/Xilinx/embeddedsw/blob/master/ThirdParty/bsp/freertos10_xilinx/examples/freertos_intr_example.c. It doesn't work, and I'm not the first to notice: https://forum.digilent.com/topic/24684-zynq-ultrascale-example-with-freertos-using-interrupts-on-a-uart-or-other-interrupt-source/

All my TTCs are enabled in Vivado. The baremetal example (https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/ttcps/examples/xttcps_intr_example.c) _does_ work, suggesting there is a problem with how the GIC and FreeRTOS are integrated.

anirudha1977, 2023-03-09:
Thanks for pointing out on this. Let me try it here and get back.

regards,
Anirudha

chris-laplante, 2023-03-24:
> Based on my observations, I see that if xiltimer is disabled and I try it, it works. I am checking why with xiltimer enabled, it might fail.

Here's a possible cause I hit on another project: When xiltimer is enabled, the port uses xiltimer to provide the FreeRTOS tick interrupt. But if xiltimer's interval_timer is left as 'none' (which is the default), the interval timer is never configured and it seems no ticks are ever generated. I think there needs to be a sanity check so if that if FreeRTOS is in use, and xiltimer is enabled, then interval_timer must be configured and enabled.

chris-laplante, 2023-03-24:
Edited previous comment to clarify. I think the problem is with 'en_interval_timer', not 'interval_timer'.

anirudha1977, 2023-04-24:
We have raised a bug request and it will be fixed in the next 2023.2 release. We will also post information abt the patch in this forum/discussion thread to get your review comments.

anirudha1977, 2023-07-20:
The issue has been fixed and will be available in 2023.2 release.