Xilinx/embeddedsw #254: Misplaced ifdef guard in xil_cache leads to endless loop

hellow554, 2023-06-22:
https://github.com/Xilinx/embeddedsw/blob/8fca1ac929453ba06613b5417141483b4c2d8cf3/lib/bsp/standalone/src/arm/cortexa9/xil_cache.c#L352-L359

there is the `#ifdef USE_AMP` inside the `while` loop which means, that it is an endless loop.
My suggestion would be to include the `while` "header" inside the `#ifdef` guard as well.

```c
#ifndef USE_AMP
while (tempadr < endaddr) {
    /* Invalidate L2 cache line */
    *L2CCOffset = tempadr;
    Xil_L2CacheSync();
    tempadr += cacheline;
}
#endif
```

Would that be correct?

anirudha1977, 2023-06-22:
Hi,
Thanks for reporting out on this.
Looks like a bug.
We will get this fixed soon In our next release.

regards

anirudha1977, 2023-08-22:
Hi,
Sorry for the delay.
The patch for this is being sent out today.

Will be available in 2023.2 builds.

regards,
Anirudha