Xilinx/embeddedsw #256: Wrong init sequence in xqspipsu_generic_flash_polled_example.c for Spansion

RandoDev2, 2023-06-29:
In file _xqspipsu_generic_flash_polled_example.c_, in function _FlashEnterExit4BAddMode()_, under the _case SPANSION_ID_BYTE0:_ clause, the source code initializes _FlashMsg[1]_ only partially then somehow switches to _FlashMsg[2]_ even though the 3rd _FlashMsg_ isn't used:

```
		FlashMsg[0].TxBfrPtr = &WriteBuffer[0];
		FlashMsg[0].RxBfrPtr = NULL;
		FlashMsg[0].BusWidth = XQSPIPSU_SELECT_MODE_SPI;
		FlashMsg[0].Flags = XQSPIPSU_MSG_FLAG_TX;
		FlashMsg[0].ByteCount = 1;
		FlashMsg[1].TxBfrPtr = &WriteBuffer[1];
		FlashMsg[2].RxBfrPtr = NULL;
		FlashMsg[2].BusWidth = XQSPIPSU_SELECT_MODE_SPI;
		FlashMsg[2].Flags = XQSPIPSU_MSG_FLAG_TX;
		FlashMsg[2].ByteCount = 1;

		Status = XQspiPsu_PolledTransfer(QspiPsuPtr, FlashMsg, 2);
```

Amit-Kumar-Mahapatra, 2023-06-30:
Hello RandoDev2,

Thank you for pointing this out. We will fix this in the upcoming release.

Regards,
Amit

Amit-Kumar-Mahapatra, 2023-11-20:
Hello @RandoDev2,

The latest release has addressed and resolved this issue.

Regards,
Amit