Xilinx/embeddedsw #257: XMutex not found in Vitis 2023.01

thymbahutymba, 2023-07-13:
Hi all, I'm trying to write a multi-core baremetal application that requires mutex for accessing shared data. However, I'm not able to import "xmutex.h". I also tried to add both libmetal library and -lmetal to GCC linker setting. What am I missing?

anirudha1977, 2023-07-14:
Hi,
Are you referring to: xmutex.h that is there at : .. /XilinxProcessorIPLib/drivers/mutex/src/xmutex.h?
If so, the design on which you are trying, does it have the AMD Xilinx Mutex IP?

thymbahutymba, 2023-07-14:
Hi,

> Are you referring to: xmutex.h that is there at : .. /XilinxProcessorIPLib/drivers/mutex/src/xmutex.h? 

Yes

> If so, the design on which you are trying, does it have the AMD Xilinx Mutex IP?

I'm not sure, how can I check it? I just created the platform using the zcu102.xsa file already available in Vitis, then I added the application in which I would like to be able to use the xmutex.h. 

I actually managed to use the metal mutex/spinlock.

thymbahutymba, 2023-07-31:
Hi @anirudha1977, looking at [system software mutexes](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/2151907329/System+Software+Mutexes+On+Versal+And+Zynq+UltraScale) I was not able to find any information on the base address of the mutex for the zynq ultrascale+. The TRM also has a broken reference to the "System Software Mutexes on Zynq UltraScale". Do you know how to retrieve the correct information on how to use system software mutex properly?