Xilinx/embeddedsw #270: xilffs Hard coded values that overide platform driver configuration in ffconf.h

cacrippen, 2023-10-13:
at the top I am guessing lines 38 - 44 are used to pull in information from the menu selections in the BSP settings menu.  we were using f_puts and f_gets in our code and trying to upgrade to the latest tools was a bit of a struggle. 
#if FILE_SYSTEM_USE_STRFUNC == 0
#define	FF_USE_STRFUNC	0	/* 0:Disable */
#elif FILE_SYSTEM_USE_STRFUNC == 1
#define	FF_USE_STRFUNC	1	/* 1:Enable */
#elif FILE_SYSTEM_USE_STRFUNC == 2
#define	FF_USE_STRFUNC	2	/* 2:Enable */
#endif

and these  defines overwrite whatever you set. 
lines  90 
#define FF_USE_STRFUNC	0

you might ask, 
Q: "why not jsut roll back to an older version since there are so many in the install folder."
[image]

A: becasue vitis will not let me!
[image]
[image]

Dear Xilinx, please stop breaking your tools and give an EASY method for selecting a different version of a driver. . . that actually works.  

How can I change ffconf.h so that it doesn't just keep getting over written each time I bring in a new HW Description and reset my BSP's? I am relatively newish to the Xilinx toolsets and there are copies of files everywhere and it all seems rather redundant and difficult to develop anything at times.

saikrishnapotthuri, 2023-10-13:
Hi @cacrippen - We recently migrated to FATFS 0.15 version, that is the reason you might be noticed so many code changes in the XILFFS 5.0 version.
Easy and right method is to remove the below line #90 in the ffconf.h file. 
#define FF_USE_STRFUNC	0
This line should not be there as this configuration will come from the BSP selection. This is a bug in the latest XILFFS version and will fix in the 2024.1 release version of the library.

Coming to your another question about using the older version of the library in Vitis, you need to change the state of the corresponding library version to "active" from "deprecated"(default state for older versions) in the mld file (xilffs_v*_*->data->xilffs.mld) of the XILFFS library like shown below.
Existing line for older versions:
OPTION library_state = "deprecated";
Updated line to use that version of the library
OPTION library_state = "active";

By doing the above change you will be able to see the corresponding library version in the drop down in Vitis tool.

cacrippen, 2023-10-23:
Thank you for your prompt reply.
I appreciate the workaround for the library version, tho it would be great if the Xilinx tool source code / files did not require modification to select a previouse version.

I will leave this open since it is goign to still be an issue till 2024.1