Xilinx/embeddedsw #274: [xilpki] Potential incorrect command descriptor for P384 private key generation

jcorbier, 2023-11-29:
Hi,

At [line 98](https://github.com/Xilinx/embeddedsw/blob/3728f546f178a1bcd91cf6efc9f8921447846cec/lib/sw_services/xilpki/src/xilpki.c#L98C55-L98C55) in xilpki.c:

```
#define PKI_ECC_NIST_P192_KEY_PRIV_GEN_CMD	0x00001701U
#define PKI_ECC_NIST_P256_KEY_PRIV_GEN_CMD	0x00001F01U
#define PKI_ECC_NIST_P384_KEY_PRIV_GEN_CMD	0x00002F02U
#define PKI_ECC_NIST_P521_KEY_PRIV_GEN_CMD	0x00004101U
```

Shouldn't `PKI_ECC_NIST_P384_KEY_PRIV_GEN_CMD` be using the same 0x01 operation ID as the commands for the other curves?

```diff
 #define PKI_ECC_NIST_P192_KEY_PRIV_GEN_CMD	0x00001701U
 #define PKI_ECC_NIST_P256_KEY_PRIV_GEN_CMD	0x00001F01U
-#define PKI_ECC_NIST_P384_KEY_PRIV_GEN_CMD	0x00002F02U
+#define PKI_ECC_NIST_P384_KEY_PRIV_GEN_CMD	0x00002F01U
 #define PKI_ECC_NIST_P521_KEY_PRIV_GEN_CMD	0x00004101U
```

sivadur, 2023-11-29:
@jcorbier  Thanks for reporting, we will check and get back on this.

jcorbier, 2023-12-07:
@NavaKishoreManne Indeed, this is currently a very minor issue when it comes to key generation. I just wanted to point it out in case this descriptor was used for actual modular addition for some other operation. Thanks for your feedback.