Xilinx/embeddedsw #293: Several files will not compile for C++

PCrabbeIMMT, 2024-03-21:
When attempting to compile our project in C++ against the embeddedsw repo, there are several headers that cause build failures. The ones we have run into are as follows:

- [xdp.h](https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/dp14/src/xdp.h): Missing ``extern "C" {}`` 
- [xdp_hw.h](https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/dp14/src/xdp_hw.h#L43): Missing ``extern "C" {}`` 
- [xv_tpg.h](https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/v_tpg/src/xv_tpg.h#L219): _Bool is not defined, had to include stdbool.h to solve this. Shouldnt this just match what other functions have done (use u8?).
- [i2stx_chsts.h](https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/i2stx/src/xi2stx_chsts.h): Missing the closing brace for the extern "C"
- [i2srx_chsts.h](https://github.com/Xilinx/embeddedsw/blob/master/XilinxProcessorIPLib/drivers/i2srx/src/xi2srx_chsts.h#L183): Missing the closing brace for the extern "C"

We have patched it on our end, but it would be greatly appreciated for these to be fixed upstream.

keryell, 2024-03-21:
Too bad.
Could you send a PR?

MajorArkwolf, 2024-03-22:
> Too bad. Could you send a PR?

The PR template says "Please do not submit a Pull Request via github. Our project makes use of mailing lists for patch submission and review.", I can submit a patch unless the maintainer specifically wants a PR?

keryell, 2024-03-22:
Interesting.
I will investigate internally about this non-PR story. @tejus26

tejus26, 2024-03-22:
@msreeram-xilinx , @saddepal or @vsrik should be able to help.

MajorArkwolf, 2024-03-22:
I do have the changes for the two issues I have raised and I can point them to this repo if you guys would like me  to.

See: 
https://github.com/MajorArkwolf/embeddedsw/pull/1
https://github.com/MajorArkwolf/embeddedsw/pull/2
https://github.com/MajorArkwolf/embeddedsw/pull/3

msreeram-xilinx, 2024-03-22:
hi @saddepal 
can you please forward to respective to check the issue.

PCrabbeIMMT, 2024-03-22:
Note that I also raised another issue for build failures in C++ as well: https://github.com/Xilinx/embeddedsw/issues/294

vsrik, 2024-03-22:
Yes @PCrabbeIMMT  - we will take a look at #293 and #294

keryell, 2024-03-22:
> Yes @PCrabbeIMMT - we will take a look at #293 and #294

Perhaps this is a good opportunity to experiment a move to a more inclusive open-source workflow based only on GitHub instead of using Jurassic mail-based patches for simple changes without IP issues?

PCrabbeIMMT, 2024-03-25:
I have emailed 3 patches to resolve #293 and #294