Xilinx/embeddedsw #298: axivdma driver is missing Example

Bethexone, 2024-05-13:
vitis 2023.2  vitis unifed IDE
在C:\Xilinx\Vitis\2023.2\data\embeddedsw\XilinxProcessorIPLib\drivers\axivdma_v6_14\examples\readme.txt中明确提供了3个例程文件，但是在vitis unifed IDE中只显示了一个
XilinxProcessorIPLib/drivers/axivdma/examples/CMakeLists.txt

Coding-xinxiangwanji2017, 2024-05-13:
Hi，您好：    我收到了您邮件，等我看完邮件以后，第一时间给您回复。

Bethexone, 2024-05-13:
[image]

HariniKatakamX, 2024-05-14:
Hi @Bethexone There are only two examples for VDMA - selftest and interrupt. The former is visible in your screenshot. The latter requires an interrupt to be connected in the design. Can you please check your design to make sure a valid interrupt is connected to GIC or INTC controller?

Bethexone, 2024-05-14:
thank，You're right. I saw the problem

Be the one
***@***.***

closed 2024-05-16 by Bethexone