Xilinx/embeddedsw #299: XUartNs550_UpdateStats should be exposed in a public header.

dhylands, 2024-05-30:
Currently, if I want to call XUartNs550_GetLineStatusReg I need to duplicate the XUartNs550_UpdateStats functionality otherwise I may lose things like framing errors (since the act of calling XUartNs550_GetLineStatusReg zeros the bits that UpdateStats looks for.

Currently calling XUartNs550_IsReceiveData, XUartNs550_IsTransmitEmpty, XUartNs550_SendBuffer, XUartNs550_ReceiveBuffer have code paths which call XUartNs550_GetLineStatusReg and don't call XUartNs550_UpdateStats.

So it would be good if either these could be fixed, or that XUartNs550_UpdateStats could be exported as a non-internal function do I don't need to duplicate it.

Coding-xinxiangwanji2017, 2024-05-30:
Hi，您好：    我收到了您邮件，等我看完邮件以后，第一时间给您回复。

ManikantaGuntupalli, 2024-06-28:
Fix will be available in next release(2024.2)