Xilinx/embeddedsw #303: MCAP cannot configure multiple devices on the same chassis

boanergies, 2024-08-19:
Hello I have a chassis with multiple Xilinx FPGAs. We use mcap to load the bitstreams into them. What we found though is mcap will find all the fpgas but only configure one of them as shown in your code:

[mcap lib query](https://github.com/Xilinx/embeddedsw/blob/5688620af40994a0012ef5db3c873e1de3f20e9f/mcap/linux/mcap_lib.c#L379)

We can fork for now and work out a solution for now but was curious if there is some use of mcap we aren't aware of that might fix this issue.

anirudha1977, 2024-09-03:
Hi,
At this point, I do not have an answer to your query. We will keep exploring on this.
Sorry abt it.

regards

boanergies, 2024-09-03:
Our fork essentially just makes sure that for all the devices found they get configured with the same bitstream provided. That for loop that I linked essentially shows it will find all the devices but your code only ends up configuring the last one that was queried. I would think you would just have to do something similar for your own code

yberthier, 2025-02-18:
Hi,
Any updates on this issue? We have several FPGAs on our infrastructure and we would like to configure these devices either by reprogramming them all at once, or find a way to explicitly target one FPGA at a time.

anirudha1977, 2025-02-18:
Hi..
Thanks for getting back on this. We did not get time to look at it. But please free to send the patch to git@amd.com