Xilinx/embeddedsw #305: msleep has no function prototype

digitalnumeric, 2024-08-30:
msleep() is listed as supported by XilTimer, but its prototype is missing from sleep.h. 
[Standalone Library Documentation: BSP and Libraries Document Collection (UG643)](https://docs.amd.com/r/en-US/oslib_rm/msleep)

XilTimer.c has a msleep() function: [xiltimer.c on Github](https://github.com/Xilinx/embeddedsw/blob/master/lib/sw_services/xiltimer/src/xiltimer.c)
`void msleep(unsigned long mseconds) {
    XilTimer_Sleep(mseconds, XTIMER_DELAY_MSEC);
}`

Prototype missing from XilTimer's sleep.h (as well as other sleep.h files elsewhere):
[XilTimer's sleep.h on Github](https://github.com/Xilinx/embeddedsw/blob/master/lib/sw_services/xiltimer/src/sleep.h)

Reference to where this was reported on the forums earlier this year:
[Vitis Unified msleep() function missing from sleep.h but present in xiltimer.c](https://support.xilinx.com/s/question/0D54U00008CPFsASAX/vitis-unified-msleep-function-missing-from-sleeph-but-present-in-xiltimerc?language=en_US)

anirudha1977, 2024-09-02:
Thanks.
We will make the corrections.

digitalnumeric, 2024-11-22:
I just checked sleep.h in Vitis 2024.2, there is still no prototype for msleep()

digitalnumeric, 2024-11-23:
Since this wasn't fixed in Vitis 2024.2, a new issue was added, this time with a patch file, to bring this issue back to the forefront. 
https://github.com/Xilinx/embeddedsw/issues/324

closed 2024-11-23 by digitalnumeric

anirudha1977, 2024-11-25:
Thanks for tracking this.
Sorry, we will fix this in the next release.

sivadur, 2024-11-27:
> I just checked sleep.h in Vitis 2024.2, there is still no prototype for msleep()
@digitalnumeric 
It is there in 2024.2 xiltimer sleep.h at https://github.com/Xilinx/embeddedsw/blob/xlnx_rel_v2024.2/lib/sw_services/xiltimer/src/sleep.h#L82
Iam not sure on which file you have checked. 
This prototype isn't needed in common sleep.h as this API is only part of xiltimer and doesn't exist in non xiltimer case.

digitalnumeric, 2024-11-27:
My apologies, I see now that you are correct.   We were looking at the wrong, sleep.h.