Xilinx/embeddedsw #306: GetPriorityTriggerType missing IntrId to IntrNum transform

digitalnumeric, 2024-08-30:
In xinterrupt_wrap.c, the function XSetPriorityTriggerType() includes code to transform the IntrId passed in as an argument to a IntrNum which it then passes to the XScuGic_SetPriorityTriggerType() function.

The function XScuGic_SetPriorityTriggerType() is missing this transform.  It takes the IntrId passed in as an argument and passes it directly to XScuGic_GetPriorityTriggerType() which results in hitting an assertion. 

This can be corrected by adding the transformation from IntrId to IntrNum into XScuGic_GetPriorityTriggerType() as done in the attached patch file:
[0001-updated-XGetPriorityTriggerType-with-IntrId-to-IntrN.patch](https://github.com/user-attachments/files/16821855/0001-updated-XGetPriorityTriggerType-with-IntrId-to-IntrN.patch)

Reference to where this was reported on the forums earlier this year:
[Vitis Unified IDE Bug in xinterrupt_wrap.h|.c XGetPriorityTriggerType() hanging at assertion on IntrId](https://support.xilinx.com/s/question/0D54U000089hXcVSAU/vitis-unified-ide-bug-in-xinterruptwraphc-xgetprioritytriggertype-hanging-at-assertion-on-intrid?language=en_US)

anirudha1977, 2024-09-02:
Hi,
Thanks.
We will take a look at it and fix it as needed.
Will use the patch.

regards

digitalnumeric, 2024-11-22:
It looks like xinterrupt_wrap.c XGetPriorityTriggerType() has been fixed in Vitis 2024.2.
I see the following comment in the code. 
" 9.2   ...  11/09/24 Update XGetPriorityTriggerType() with IntrId to IntrNum transformation."

When will the github repo be updated to reflect the state of the code matching the 2024.2 release?

digitalnumeric, 2024-12-07:
See previous comment.

closed 2024-12-07 by digitalnumeric