Xilinx/embeddedsw #307: Support FreeRTOS SMP in future release

digitalnumeric, 2024-09-05:
Since AMP is not supported for the Zynq UltraScale+ APU (A53-0 thru A53-3), it would be useful if FreeRTOS with SMP was supported.  

According to this link: [FreeRTOS SMP on Github](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/smp), SMP support has been merged into the main FreeRTOS-Kernel beginning with FreeRTOS v11.

 Quote from the link above: 
"FreeRTOS SMP feature has been merged into FreeRTOS-Kernel main branch. We recommend you to use the FreeRTOS-Kernel main branch to develop FreeRTOS SMP applications.”

From the FreeRTOS web-site: [Symmetric Multiprocessing (SMP) with FreeRTOS](https://www.freertos.org/Documentation/02-Kernel/02-Kernel-features/13-Symmetric-multiprocessing-introduction)

anirudha1977, 2024-09-06:
Hi,
We dont have any plans to officially support FreeRTOS SMP for various reasons and our marketing requirements.

regards,

digitalnumeric, 2024-10-04:
> Hi, We dont have any plans to officially support FreeRTOS SMP for various reasons and our marketing requirements.
> 
> regards,

If not FreeRTOS with SMP, has the team considered supporting another RTOS such as RTEMS, ThreadX or maybe Zephyr?

anirudha1977, 2024-10-05:
We are moving to Zephyr.

digitalnumeric, 2024-10-07:
Good to know.   Any idea when Zephyr support will first appear in Vitis so that we can prepare?

Will the current level of FreeRTOS support be retained when Zephyr support begins?

anirudha1977, 2024-10-08:
We are working on the Zephyr support in Vitis. No date is finalized yet.
Zephyr will be available through its own AMD github repo. Users can use Zephyr west build system to build Zephyr.
Also, we will add support through west, where given an xsa, west will invoke SDT, Lopper to create device tree needed for Zephyr.

For future AMD generation projects, FreeRTOS will not be supported. The existing FreeRTOS support in its current form will continue for some more time.

digitalnumeric, 2024-10-09:
Will west be integrated into Vitis or will it only be used from the command line?

Currently the Zynq UltraScale+ MPSoC VCU is only supported on Linux with unofficial (non supported) here:
https://github.com/Xilinx/vcu-ctrl-sw

Is there any chance that the VCU might be supported in Zephyr projects?

digitalnumeric, 2024-11-23:
Right now on the Zephyr web-site I only see KV260 support for the Cortex-R5.  I am specifically interested in SMP for the Zynq UltraScale+ MPSoC APU (the four A53 cores).   Are there plans to support the APU with Zephyr ?

anirudha1977, 2024-11-25:
Hi,
SMP for ZU+ + MPSoC will take at least 2 releases. May be we can have EA for 2025.2 release. Last quarter of CY25.

digitalnumeric, 2024-11-25:
I understand.  That makes sense.   

When you say SMP are you saying the APU with the four A53 cores will be supported in that time frame?

closed 2024-12-28 by digitalnumeric

digitalnumeric, 2025-02-19:
I don't see much activity on the AMD Zephyr project.

Are they still targeting the end of 2025 for a first Vitis release with SMP support for the APU with the four A53 cores?