Xilinx/embeddedsw #309: Unsupported/invalid standard USB standard device requests should stall EP0 (usbps/usbpsu)

Minimal-C, 2024-09-15:
## Background

Discovered this when running a usbpsu example and running an ancient version of "uvcview.exe" (6.1.7600.16385).

When running "uvcview.exe" (6.1.7600.16385) the example device enters a bad state and the device cannot be used until restarted.
When running a more recent "usbview.exe" (10.0.17134.1) the USB descriptors are displayed correctly.

The old "uvcview" appears to enumerate over string descriptors 0-31 regardless of what string descriptor indexes are referenced in other descriptors.
The new "usbview" appears to only request string descriptors that are actually referenced in other descriptors.

I believe the issue is that some of the Xilinx USB drivers don't stall on unsupported/invalid standard descriptor requests when they should.

To attempt to verify this I implemented a small local patch to stall when the host requests an invalid string descriptor id and this appears to have resolved the issue.

### The USB spec

> If an unsupported or invalid request is made to a device, the device responds by returning a
STALL Transaction Packet
> [Universal Serial Bus 3.2 Specification, Section 9.4, page 332]

## Affected drivers

Note: I have only ran the usbpsu driver on hardware but suspect the issue is also present in the usbps driver.

|  | usb | usbps | usbpsu |
|-|-----|--------|---------|
| Stalls?  | Yes | No | No |

The infrastructure for stalling appears to already be present in usbps and usbpsu, but is not active for all standard descriptors.

## Expected Behavior

If an unsupported or invalid request is made to a device, the device responds by stalling EP0.

## Actual Behavior

In usbpsu, if an unsupported or invalid string descriptor request is made to a device, the device responds by sending a packet of length 0:
https://github.com/Xilinx/embeddedsw/blob/5688620af40994a0012ef5db3c873e1de3f20e9f/XilinxProcessorIPLib/drivers/usbpsu/examples/xusb_ch9.c#L292C1-L305C31

## Steps to Reproduce

1. Run the usbpsu/xusb_ch9_storage example
2. On the host PC run uvcview.exe (6.1.7600.16385)

radheyxilinx, 2024-12-19:
@Minimal-C thanks for reporting the issue.  We have an answer record for it.  Also i have created an internal ticket to track the changes submission to the next public release.https://adaptivesupport.amd.com/s/article/71925?language=en_US

Minimal-C, 2026-10-26:
Closing: This was resolved by 9f7889ec8671ca75b296820e6aab599fc1205ba7

closed 2026-10-26 by Minimal-C