Xilinx/embeddedsw #317: Vitis Unified add support for sub-folders under src folder

digitalnumeric, 2024-10-04:
To allow organizing the many files making up a project, please add support for creating sub-folders under the src folder.

anirudha1977, 2024-10-05:
Hi,
Can you please provide more context?

Regards

digitalnumeric, 2024-10-07:
This unanswered issue on the forums provides more context:
Vitis Unified CMakeList.txt collect () error with backslash in sub-folder path
https://adaptivesupport.amd.com/s/question/0D54U00008GmixqSAB/vitis-unified-cmakelisttxt-collect-error-with-backslash-in-subfolder-path

digitalnumeric, 2024-11-22:
Here is another forum post for reference:
https://adaptivesupport.amd.com/s/question/0D54U000089goOhSAI/how-can-i-get-vitis-20232-to-compile-source-files-in-subdirectories?language=en_US