Xilinx/embeddedsw #318: Vitis Unified platform build fails when top-level lwip_debug=true

digitalnumeric, 2024-10-04:
In the platform, trying to turn on lwip debug print statements for sub-systems (e.g., socket, udp, etc.) results in nothing is printed.

If I try to turn on the top-level lwip debug the platform build fails. 

Is there a work around to turn on lwip debugging in Vitis Unified?

anirudha1977, 2024-10-07:
Hi,
We will check and get back.

VenkateshOdela, 2024-10-07:
@digitalnumeric , the sub-system socket or udp level debug messages are shown when top-level lwip_debug is enabled.

digitalnumeric, 2024-10-07:
Attached are screenshots with the top-level lwip debug set to false and true.  With it set to false the build succeeds.  Setting it to true the platform build fails. I also attached the text from the Vitis Unified Output Window. 
[image]
[image]
[platform_build_output_error.txt](https://github.com/user-attachments/files/17280003/platform_build_output_error.txt)

digitalnumeric, 2024-10-07:
Update.   I found the following fix and applied it to lwip213
lwip220 : fix lwip debug options in cmake
https://github.com/Xilinx/embeddedsw/commit/1fee17e71664a082382099049295bfb9abdc0426

[0001-applied-fix-to-allow-selection-of-SocketAPI-with-Fre.patch](https://github.com/user-attachments/files/17281775/0001-applied-fix-to-allow-selection-of-SocketAPI-with-Fre.patch)

However, the platform is still failing to build.  One error I see in the logs as the tool cannot find xil_printf.   Vitis logs attached.
[2024-10-07_ide.log](https://github.com/user-attachments/files/17281830/2024-10-07_ide.log)
[2024-10-07_ide_verbose.log](https://github.com/user-attachments/files/17281831/2024-10-07_ide_verbose.log)
[2024-10-07_vitis.log](https://github.com/user-attachments/files/17281832/2024-10-07_vitis.log)

BTW: We are still on Vitis Unified 2023.2 because we found 2024.1 much too unstable.  We were unable to build the same project (or even simple projects) and after fighting with it for 2 days had to give up.  I posted about this on the forums and informed our FAE.    We are going to have to wait until 2024.2 comes out before trying again.

venodela, 2024-12-19:
Hi @digitalnumeric 

As per the Vitis log, we resolved the build failure issue caused by the NETIF_DEBUG macro in the 2024.2 release.
@anirudha1977 @HariniKatakamX

digitalnumeric, 2024-12-19:
Sounds good.  Unfortunately due to other issues with 2024.2 we are unable to use it.

closed 2024-12-19 by digitalnumeric

anirudha1977, 2024-12-19:
Hi,
Sorry to know about it. We will try our best to ensure that 2024.2 is usable.
These issues that you are referring to, are these already mentioned in this thread?

If so, we will go oved them once again and see what we can do.

digitalnumeric, 2024-12-19:
I just added the issues this morning.  Please see these:
Vitis 2024.2 is missing CMake definition for \_\_FILENAME\_\_ used in xil_assert.h in some cases
https://github.com/Xilinx/embeddedsw/issues/326
Vitis 2024.2: CMake Policy CMP0010 is not set: Bad variable reference syntax is an error.
https://github.com/Xilinx/embeddedsw/issues/327
Vitis 2024.2: Fails to complete build, hangs
https://github.com/Xilinx/embeddedsw/issues/328

jwolfm98, 2025-03-27:
I saw a few of your threads now - a lot of the same issues I've run into.

We fixed a lot of this by switching to using Classic instead of Unified.

My experience with Vitis 2023.2+ in the last 10 months has been very rough, there are _still_ a lot of problems with unified.

digitalnumeric, 2025-03-28:
We were using Vitis Classic and we were having so much trouble, especially with custom VHDL IP, that we switched to Vitis Unified 2023.2.  It was an improvement over what we had been dealing with.

Unfortunately, we have been unable to switch to a newer Vitis Unified because of all the new issues introduced.