Xilinx/embeddedsw #320: Vitis Unified remember open files between sessions

digitalnumeric, 2024-10-10:
Right now when you close and reopen Vitis you are always presented with a blank slate.   It does not remember the files/tabs you had opened previously.  

Modern IDEs typically remember the files/tabs from the last session the project was opened.

Update Vitis Unified to remember open files/tabs between sessions so that when reopening a project it automatically opens all the files/tabs that were opened when Vitis was last closed.

sdxtest1, 2024-12-19:
Hi @digitalnumeric ,
Its working fine with 2024.2 Vitis unified. After launching the Vitis with existing workspace then all previously opened tabs are available in editor view.

Which version of tool and OS you are using?

digitalnumeric, 2024-12-19:
We were originally using 2023.2.  We checked checked this on 2024.2 and you are correct it is working there.

Unfortunately we are unable to make use of Vitis 2024.2 because of these issues below:
Vitis 2024.2 is missing CMake definition for \_\_FILENAME\_\_ used in xil_assert.h in some cases
https://github.com/Xilinx/embeddedsw/issues/326
Vitis 2024.2: CMake Policy CMP0010 is not set: Bad variable reference syntax is an error.
https://github.com/Xilinx/embeddedsw/issues/327
Vitis 2024.2: Fails to complete build, hangs
https://github.com/Xilinx/embeddedsw/issues/328

closed 2024-12-19 by digitalnumeric